opencardev/crankshaft #83: Audio Crackles and Crackles Even Worse After "Hey Google" Voice Command

chaoman, 2018-04-24:
- What phone do you have? What OS version? 

HTC 10 running Android 8.0. I am using an external USB Creative SoundBlaster Play! 2. 

- Did you try to enable and run autoapp under X11 dev mode?  Follow this: https://github.com/htruong/crankshaft/wiki/Crankshaft-dev-mode

Yes, got into the dev mode and made changes 2 Pulseaudio config files according to issue #50 

`/etc/pulse/daemon.conf`

`default-fragments = 10`
`default-fragment-size-msec = 10`

`etc/pulse/default.pa`

removed `tsched=0`
and
changed `set-sink-volume 0 32428`

- Please provide any further information that you might find helpful.

I am running the latest release Crankshaft Openauto 0.2.3 2018-04-18 and the audio problem seemed to have gotten worse. I reported #50 earlier and the audio playback crackle issue was alleviated. However, with the latest build, despite the changes applied in #50, the audio still crackles. It will be worse after I enabled the Hey Google voice command. This all happens when I have the audio out from the connected external USB Soundblaster Play! 2. 

With the Raspberry Pi's built-in audio jack, the crackle didn't happen. I am guessing it has something to do with the external USB sound card.

Vitalic66, 2018-04-24:
Nearly the same for me, except i don't use crankshaft but the original openauto on which crankshaft is based. I got a hifiberry dac and had same crackles you described. Only way so far to get good sound is to use bluetooth for audio instead of usb stream. This should fix your problems too.

poxin13, 2018-04-25:
Ended up having the issue as well with a HifiBerry. Once you run an Ok Google command, the sample rate switches to 16khz and messes up pulseaudio. No audio will be heard if you force 48khz as voice doesn't run at that sample rate. I ended up removing the HifiBerry for now and using a USB DAC. No issues now.

poxin13, 2018-04-25:
I modified it a little bit with the USB DAC, from memory something like these changes:
```
high-priority = yes
realtime-scheduling = yes
exit-idle-time = -1
resample-method = ffmpeg
default-sample-rate = 48000
```

chaoman, 2018-04-26:
Tried the above settings and also referenced the wiki in openauto but still no luck after "OK Google" command. However, the audio only part stopped crackling! At least I am happy for now without the voice command.

chaoman, 2018-05-12:
The key to alleviate the audio output crackle is to edit /etc/pulse/daemon.conf and comment out everything but the following line:

`resample-method = ffmpeg`

Audio output quality via USB DAC is superb but I could still hear some static from "OK Google" voice command.

Maju3, 2018-06-01:
Hmmm i am just wondering when you comment out everything does pulse audio work?
How does pulse audio know the default sample rate then ?

Maju3, 2018-06-28:
@hawkeyexp
What audio device do you use?
Hifiberry or USB dac/soundcard?

hawkeyexp, 2018-06-29:
I have a hifiberry dac plus and external usb mic - alternate internal rpi audio. Both sounds better without the crackle.

tehniemer, 2018-06-29:
What are your settings for the audio?

hawkeyexp, 2018-06-29:
if have injected this settings: [https://github.com/opencardev/crankshaft/blob/crankshaft-ng/stage2/03-crankshaft-base/files/etc/pulse/pulseaudio_daemon.conf](url)

Maju3, 2018-06-29:
Ok what do you use as audio backend?
I just ask cause after switching to QT AUDIO all crackles are gone for me.
I use a USB DAC.

The setting to switch between RT audio or QT audio is on the crankshaft mainscreen in tab "audio"

Maju3, 2018-07-01:
@hawkeyexp 
I use the same pulseaudio Settings like you:
default-sample-rate = 48000
resample-method = soxr-hq
default-fragments = 8
default-fragment-size-msec = 5

So i can confirm that this Settings work great!
As Audio Output backend i use QT Audio.
No crackles on Music Playback or Navigation voice!
Only SOMETIMES i get a humming Sound.This seems to come up when the pi is working(cpu or Access to sd Card).
I think that is related to usb Audio in General.
Maybe in near future i switch to a hifiberry dac and replace my usb dac with it.

Hisma, 2018-07-04:
I also noticed I have this problem.  I will try settings suggested by hawkeye and see if it works as I have a similar audio setup.  Just for extra precaution, where must these settings be applied - /etc/pulse/daemon.conf?

hawkeyexp, 2018-07-04:
yes - `/etc/pulse/daemon.conf` is correct

hawkeyexp, 2018-07-09:
As Maju3 also reported the pulse settings used in CS-NG are fixing the problem so closing issue.

closed 2018-07-09 by hawkeyexp