opencardev/crankshaft #84: Full build from source - problems with qt5 and openauto

hawkeyexp, 2018-04-25:
I have done a full build like you described in your compile info.
i noticed that openauto won't build if the qt5 was built before caused by missing bluetooth qtmultimedia etc.
After some tries and a make uninstall of qt5 the build was possible.
I think the build instructions are not fully correct :-)

htruong, 2018-04-26:
Oh that's right. Since you have compiled qt5 manually, installing qt5lib etc will cause it to be confused. Let me change that :)

hawkeyexp, 2018-04-26:
I could solve the bug by removing all possible existing qt5 packages which was installed via apt.
Next time i will release a set of scripts which will make it possible to build all needed parts of crankshaft from sources by an easy way :-)