opencardev/crankshaft #85: Separate volume control navigation voice/music

rubenverhoef, 2018-04-25:
I would like to have independent volume control of the navigation voice and my music. Could this be possible? So i can set a "fixed" volume for my navigation voice, and when i turn the volume up for my music the lovely voice of my navigation will not scream at me.
Maybe this is more for the OpenAuto software side? Then we have request this there.

htruong, 2018-04-25:
Speech and Music volumes can be independently controlled, so this is possible. Do you mind filing with f1xpl OpenAuto first to see if he wants to help out? I will try to implement it if f1xl refuses it.

rubenverhoef, 2018-04-25:
Thanks for the great support, I will make an feature request at f1xpl OpenAuto.
Maybe you see also something is this: https://github.com/f1xpl/openauto/issues/64
:)

rubenverhoef, 2018-04-26:
it should be possible already? 
https://github.com/f1xpl/openauto/issues/79
So maybe we can make a gui or do it with hardware buttons to change the volumes independently

s0len0id, 2018-04-26:
That would be nice!
How does this work though? On my 7.1.1 Z2Play its the other way around: even though in normal Google Maps i've set the volume to 'high', TuneIn Radio has a way higher volume. The Google Play Music player seems to have a similar volume as the Google Maps navigation voice. Maybe yet another player app provides a level control of its own?

Or actually, does Google Maps send navigation messages over the 'notifcation' audio channel? 
(Which in the case of a bluetooth connection gets mixed together with the 'media' audio channel over the bluetooth media channel? Those can't be separated anymore ofcourse.) 
And in the case of a usb connection all audio channels remain separate in the H264 stream?

rppgen, 2019-07-31:
Folks, How to install Spotify, TuneIn on rasbt pi that has crankshaft?