opencardev/crankshaft #86: systemd rules only work in Dev Mode

techdev5521, 2018-04-26:
Phone: Google Pixel 1
OS: Android 8.1.0

I wrote a python script that listens to GPIO input and overlays a camera preview for a Pi Camera. I also wrote and successfully tested a systemd service unit to run said python script on boot. The systemd service runs in dev mode but not in non-dev mode.

I have tried running Android Auto under both EGL and X11 without success.

How can I get my systemd service to run under non-dev mode?

htruong, 2018-04-26:
It should start in non-dev mode. I think the problem here is somehow the service failed to start. One big gotcha is that everything will be mounted read-only, so anything that expects the filesystem to be read-write will probably crash.

Other than that, you should be able to SSH in non-dev mode by doing something like this.

```bash
$ sudo /opt/crankshaft/devmode.sh disable
$ sudo systemctl enable networking.service
$ sudo systemctl enable dhcpd.service
$ sudo systemctl enable avahi-daemon.service
$ sudo systemctl enable ssh
```

Then edit the `/boot/crankshaft/crankshaft_default_env.sh` to turn off dev mode. Then restart.

Then do whatever you need to do to inspect the systemd unit. Please let me know if that helps.

techdev5521, 2018-04-26:
Thank you!

Having networking without dev mode helped a lot! There was indeed an issue with the read-only filesystem. I was trying to create a log file but my script failed because it couldn't create this log. However, when re-enabling dev-mode but booting without it, it seems that the Pi camera is disabled. 

How can I keep the Pi camera enabled when not in dev-mode?

htruong, 2018-04-26:
Hmm, I don't know. At least I didn't specifically do anything to disable it. If it gets accidentally disabled, then that should be categorized as a bug. :)

closed 2018-07-09 by hawkeyexp