opencardev/crankshaft #90: /etc/customization_done and /etc/fs_read_only_config fails on second run

hawkeyexp, 2018-04-28:
I noticed following:
system boot and run crankshaft.sh all works fine and new image is configured.
if i delete crankshaft-xxxx.img to build a new one script skips customization cause it is detected as customized (check files exists). But that's wrong - mounting image shows same - no check files.
After system reboot all is fine again for one run.
I have no idea why and a check of possible still mounted paths are also ok.
Possible you have an idea...

htruong, 2018-04-28:
That's because Ubuntu is messed up :) It mounts the old image even if you deleted it from the disk. You'll need to restart the computer each time you make a new image. I know Fedora doesn't have that bug, but it has other problems.

I think 18.04 might have fixed it, but let me test & confirm it. I will let you know.

hawkeyexp, 2018-04-29:
I'm working with debian stretch or raspbian stretch and same prob...
If it's only the filename let's do a trick and add time to filename - then we have on every build a new filename and the prob should fixed.

htruong, 2018-04-30:
Guess this is resolved by yourself :)

closed 2018-04-30 by htruong