opencardev/crankshaft #91: Why would AndroidAuto launch on the *phone* ?

bhwithun, 2018-04-29:
I have a `Pixel2`, `Rpi3B+`, official `touchscreen` and Crankshaft works great.
I have another 2 other `Pixel 2`'s which also works great.

I have a `Pixel1 XL`, which does not work.
A friend of mine has a `Pixel1 XL` which also does not work.

All of these phones run Android 8.1.0.
My Rpi is running 0.2.2

I know the developer has a `Pixel1 XL`, so I'm a bit surprised by this.
Stranger still -- Often (but not always) when I plug in the `Pixel1 XL`, Android Auto launches on the PHONE rather than on the RapberryPi.  Why would that be?  Is it a clue to something that is misconfigured?

htruong, 2018-04-29:
Do you have debugging mode enabled on the P1XL? It seems like sometimes it doesn't work with debugging mode on.

bhwithun, 2018-04-29:
No I checked that specifically because I heard that it can cause problems.

I *have* had debug mode on, but it was not on at the time.

Are there any log files I can check, or enable, that would provide clues?

htruong, 2018-04-29:
Thanks so much for taking time looking at the issues :). 

I can't say anything intelligent now about the situation, to be honest. I know the changes to OpenAuto that happened in the last month was solving some issues but makes it a little bit more flaky. Please hold on, I probably will try another build next week or so to see if that fixes the issue. The fact that Android Auto launches on the phone probably just means that the Android Auto somehow detects that there is a head unit so it launches. But then it realizes it can't project, so it launches on the phone. That's my wild guess...

htruong, 2018-05-04:
Note: I just noticed I couldn't get OpenAuto to launch if I set the resolution to 720p.

vanshkejriwal7, 2018-06-03:
I have the same problem pls help. I hope it will be resolved in next software update.

DJFliX, 2019-01-19:
@bhwithun @vanshkejriwal7 are you still experiencing these issues? If so: which version of Crankshaft/Crankshaft-ng are you using?

closed 2021-11-27 by github-actions[bot]