opencardev/crankshaft #94: tocuhscreen rotate and calibration

computeridea, 2018-05-05:
i, i'm tryng cranckshaft openauto with a custom display touchscreen, it work fine in EGL mode, but the axis are inverted, how can i change it?
b.regards
Davide

chunky74, 2018-05-05:
Not sure if this will help - but I had a similar problem with an overlay touchscreen.  My fix was to remove my ribbon connector and replace it in the other way around.

computeridea, 2018-05-05:
exactly.. for start voice input i have to touch right-bottom corner in change of right-upper corner
if si move the touch horizzontal the screen rool upo and down, if i touch vertical it put away event widjet from screen

chunky74, 2018-05-05:
Could you post a link to the touchscreen you are using please ?

computeridea, 2018-05-05:
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.co.uk%2Fulk%2Fitm%2F262118994729

Vitalic66, 2018-05-05:
Had similar problems with these capacitive touch. Bought an inductive on. No setup needed anymore...

You might change xinput settings. There are several instructions how to setup eGalax touchscreens. Guess you have one too...

computeridea, 2018-05-06:
i have two touch controller,  one capacitive, but it doesn't works with raspberry or crankshaft, and one resistive, with the capacitive one it does as yous say if i cionnect directly on android, if i invert the cable, it inverts the axis, but it inverts boths axis, but what is needed is to rotate axis of 90°,  anyway the resistive one doesn't have effect.. and it's the one that works on raspberry.. 
so i think another solution is needed

computeridea, 2018-05-10:
hi guyes, thanks to you all, in the end i could calibrate the eGalax controller, througt raspian and copied the configuration on crankshaft, but i have still a problem, in the demo screen, before i connect the phone, i can follow the touch curson on the screen for each corner, but when i connect the phone, it seems like the touch doesn't work on right and on botton, i can slide the widjet on center screen, but i can't push button on botton bar, or start voice command on upper right..
.. maybe it concern something about resolution when google auto start?  any advise?

WhereisBielefeld, 2019-01-13:
Hey,
I´m struggling with the same issues already mentioned. I also have got a eGalax Controller (Sainsmart 7" Display) and got my values Touchscreen set in "/etc/X11/xorg.conf.d/99-calibration.conf" but no effect. Neither in AA nor in OpenAuto. I also varied the values but nothing changed after a reboot nor before.

Section "InputClass"
   Identifier "calibration"
   MatchProduct "eGalax Inc. USB TouchController"
   Option "Calibration" "121 1917 317 1741"
   Option "SwapAxes" "1"
EndSection

Is there a trick or did I overread something?
I started with X11 (set in Crankshaft config). Any clues?

WhereisBielefeld, 2019-01-14:
ok... i got my mistake. I hope it solves someone elses....

/usr/share/X11/xorg.conf.d/99-configuration.conf

Thats the right place to save the values!

closed 2021-11-27 by github-actions[bot]