opencardev/crankshaft #95: Feature Request - Headlight Controlled Screen Brightness

t00fatt, 2018-05-06:
Would it be possible to add a feature which would allow assigning a pin which would change the screen brightness between two user configured values on high or low inputs? This would allow connecting the headlight signal from the car to automatically control screen brightness with the headlights.

I know this is possible with a simple script in addition to crankshaft but building it into crankshaft itself would make updating more painless, and I think many would benefit from such a feature.

htruong, 2018-05-10:
Sorry I did not see your question earlier. Sure, try this GPIO solution for brightness up and down:

https://github.com/opencardev/crankshaft/wiki/Customizing-Crankshaft

Uncomment the file `/boot/crankshaft/gpio2kbd.cfg` so you have two keys defined for brightness up and down.

Then edit the file `/boot/crankshaft/triggerhappy.conf`

Instead of `brightness.sh up`, replace it with `brightness.sh up set blah` where blah is a number from 0 to 255.

Have fun!

t00fatt, 2018-05-10:
Can this be configured to work with just one input pin? So when the signal is high on that pin brightness is set to X, and when the signal is low brightness is set to Y?

htruong, 2018-05-10:
Triggerhappy seems to support it natively. I don't know how well it works,
but this should at least theoretically work.

Docs:

https://github.com/wertarbyte/triggerhappy/blob/master/README

The gpio2kbd daemon works exactly like how you would expect a keyboard key
to work. So when you press, it triggers state 1 of triggerhappy, hold -
state 2, and release - state 0. So you could map a GPIO to an abirtrary
keyboard key, let's say the BRIGHTNESSUP key.

```
BRIGHTNESSUP 1 .../brightness.sh set 200
BRIGHTNESSUP 0 .../brightness.sh set 10
```

Will probably do it.

t00fatt, 2018-05-10:
Thanks! I got it working!

How can I issue a shutdown command through triggerhappy.conf? Does it have sudo? I tried the sudo shutdown -h now command and it did not work. I need my pi to shutdown on a rising trigger, and would like to avoid writing a script just for this function.

closed 2018-05-11 by t00fatt