opencardev/crankshaft #96: Night mode doesn't appear to be working properly

dasonwong, 2018-05-07:
- What phone do you have? What OS version? Pixel 2 XL, 8.1.0

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always reproducible.
1. Plug in phone when night time
2. Android Auto shows on Crankshaft, interface is in Day mode

- Please provide any further information that you might find helpful.
On the phone I've tried both 'Car-controlled' and 'Phone-controlled' for Day/Night control, neither of them provide the desired functionality. Using the same setting on the phone forcing Night or Day works but the desired functionality is for it to change automatically based on time of day.

If I'm misunderstanding the functionality please let me know.

htruong, 2018-05-07:
Please try this: https://www.reddit.com/r/crankshaft/comments/8eze6m/heres_an_app_that_forces_android_auto_to_switch/

dasonwong, 2018-05-07:
I just read about that app and I haven't tried it yet. But I'm wondering why an extra app is necessary? Is it a limitation of Crankshaft or a bug in Android Auto?

htruong, 2018-05-07:
I think it's due to a couple of reasons: 

- The default mode - controlled by car - was impossible do implement because the Pi doesn't have a light sensor, nor does it know the time to "fake" sensory data based on time.
- I have no idea why the phone controller mode doesn't work by default. Perhaps Emil can shed some lights on it, but it's unlikely something Crankshaft could intervene.

szeater, 2018-05-09:
@htruong would it be possible to build an interface for light sensor data to be fed into crankshaft to support the 'controlled by car' method? I ask because in my situation (and potentially for others too) I would actually be able to get that information to the Pi with little to no effort.

htruong, 2018-05-09:
Yes. Although I don't have a timeline for this, but this API is already reverse engineered. Whenever possible, this will be a great start. Perhaps a GPIO to trigger or to read it from the CAN bus would be nice. 

https://github.com/f1xpl/aasdk/blob/development/aasdk_proto/NightModeData.proto

adamoutler, 2018-05-15:
my Android Auto deck does not get data from the phone.  it gets data from the car or itself.  I didn't like the way the car auto-dimmed when I turned the lights on, so I used timestamps.  Daytime start and stop.  I found a timestamp here https://github.com/opencardev/aasdk/blob/development/src/Messenger/Timestamp.cpp.  This should work for timestamp-based switching.

snailium, 2018-05-17:
@htruong Would you please detail the proto part?

I'm working on adding a BH1750FVI (I2C light sensor) to auto adjust the screen brightness. Can we feed in the BH1750FVI reading (raw or calculated value) to switch night mode?

snailium, 2018-05-17:
@adamoutler I also managed adding a DS3231 (I2C RTC) module. It seems working well. It is one way to get real time.

adamoutler, 2018-05-17:
You don't want an actual light sensor because cars drive under trees, Bridges and clouds.  Rather, a method of switching when ground or high is detected on the illumination line on the car and time.   An RTC is too much for this.  Just a sunup and sunset approximate time is how this is generally handled.  +/- ~15minute

KK1423, 2018-05-19:
I would also like some guidance on this. My project already has circuitry that connects the car's DIM line to one of the gpio, and I would appreciate it if I could have that switch between day and night mode.

KK1423, 2018-05-19:
https://github.com/opencardev/openauto/blob/355c1d3e21e96c61e5bb7fc3d6106329db405162/src/autoapp/Service/SensorService.cpp#L124
This is where openauto is hardcoded to only send a false is_night sensor reading. Modifying this to depend on other sources of information may be insightful.

hawkeyexp, 2018-06-27:
With upcoming builds based on crankshaft-ng branch system has included day/night switch via gui or time based if rtc installed. The gui state (day / night) is reflected to android auto during connect.

hawkeyexp, 2018-07-09:
Released with CS-NG

closed 2018-07-09 by hawkeyexp

adamoutler, 2018-07-09:
Not good enough.  Consumer models switch in real-time not at connection.  This is not complete.

hawkeyexp, 2018-07-09:
Yes that's correct. But they have a builtin light sensor and most users didn't have one.
The rtc solution and switch option via gui is currently the best compromiss. I keep an eye on it for later releases.

adamoutler, 2018-07-09:
I don't know how to get current lat/long but sunset calculation should be
possible with those values.

tehniemer, 2018-07-12:
The workaround app solution seems to break CSNG, when installed the interface screen flashes, but the navigation bar remains constant.