opencardev/crankshaft #98: Media Apps not showing in media tab

downraiin, 2018-05-08:
This is propably more an Android Auto problem, than a crankshaft issue, since this applies to the App aswell, so crankshaft isn't even used when this first occurs, but i will explain my problem anyway, in case anyone else experiences similar issues or knows a fix. I have an open Android Auto discussion here: https://productforums.google.com/forum/#!topic/android-auto/WHSiNComPRk;context-place=forum/android-auto

Now, to my problem:
The first time i installed Android Auto on my Phone, it worked fine. All compatible apps that i was using were available to use and functioned correctly. After connecting my phone to the pi for the first time, it worked fine aswell. Especially Spotify in my case, appeared in the media tab and was controllable. This was still at home, before using the pi in my car. So i went to my car, installed it there, and wanted to enjoy my first ride with a new headunit, but now, just a few minutes later, none of those media apps that were available before were there. Only Google Play Music was there. There was also no arrow/triangle next to the media icon, touching it again had no effect.
After trying every different possible solution i could find, i uninstalled every app i wanted to use with android auto, including android auto, reinstalled all of them and tested again: All apps were showing up in Android Auto on the phone (App, not crankshaft). Then i connected my phone to the pi again, and it worked as well. All apps were there. However, the next morning, i wanted to use the unit in my car again, and when i connected my phone in the car, it was all gone again. 
I believe that this is a problem with Anroid Auto, rather than crankshaft, because when my apps are not shown in crankshaft, they are also not shown in Android Auto on the phone, but like i said, i am posting here as well, in case anyone experiences the same or knows a fix.
If any kind of solution/workaround comes up, i will update here aswell.

- What phone do you have? What OS version? Lenovo Zuk Z2, OS version: 1.9.136 ST, Android Version 6.0.1
- Steps to reproduce the bug: I didn't reproduce it on purpose yet, but i am pretty sure, that it works right after reinstalling all apps, and will stop working after closing/disconnecting the first time.

htruong, 2018-05-09:
Thanks for the detailed report. This is the first time I have heard of this problem. I will keep tab on the issue and see if there is any other user affected and any way to rectify it.

hawkeyexp, 2018-07-12:
I know this behaviour in another situation: some time ago i tried to include my apps directly to rom (/system) - but android auto seems to don't like this and needs its apps on /data to work correctly.
I think this problem can be closed cause it's not caused by cs - it must be caused on the other side.

closed 2018-07-12 by hawkeyexp