opencardev/crankshaft #99: Question - System shutdown from rising edge trigger on pin 3

t00fatt, 2018-05-11:
I am using a NPN optocoupler to sense the state of my ignition wire. So, when the ignition is on (+12v) my optocoupler sets a pin low, and when the ignition is off(0v) the pin is pulled high. The default function on GPIO3 only triggers a shutdown from a falling eddge, but I need the trigger to shut down from a rising edge trigger. Can the shutdown trigger be configured to detect either a rising or falling edge?

I tried using the triggerhappy function, but when I call /opt/crankshaft/poweroff.sh from the triggerhappy function the system does not shutdown. I suspect this is because trigger happy executes commands as nobody and does not have sudo access. I'd like to avoid having to SSH into my pi every time I update to give nobody sudo access.

Maju3, 2018-05-11:
Why not use a relay instead of a optocoupler?
https://github.com/opencardev/crankshaft/wiki/Boot-and-Shutdown-the-Pi-with-ignition-key

htruong, 2018-05-12:
Can you try to modify triggerhappy.conf so it says:

```
WHATEVER_MAPPED_KEY 1 /opt/crankshaft/dumb_suid poweroff.sh
```

When triggering that way, it should not require further interactions from the user, so theoretically should work. I think you mentioned sudo didn't work, I am puzzled somewhat. Regardless, please try that to see if that takes it.

Triggerhappy actually should execute as pi, so let's hope that this works.

Maju3, 2018-05-12:
Ok wait.
The DEFAULT Funktion of gpio3 is BOOT from a falling edge.
But when you add "dtoverlay=gpio-shutdown" to config.txt then you get a toggle button Funktion.
That means when the PI is OFF then a falling edge on gpio3 will wake it up.
When the PI is running then a falling edge on gpio3 will shut it down.

So it is allways triggered by a falling edge.

Maju3, 2018-05-12:
BUT try that:
Run the PI from a constant power source.
Now Put a Jumper on gpio3 and let the Jumper PLUGED in.
The PI will shutdown now.
Now when the PI is fully shuted down then remove the Jumper and the PI will start up again.
So it seems like the PI does start up/shutdown even on changing edges;)

Maju3, 2018-05-12:
That is why i use a relay for this.
I connect the relay "NORMALLY CLOSE" to ignition.
This way gpio3 get a falling edge when my ignition is OFF(relay contacts Close).
And when i turn ignition ON then gpio3 gets a rising edge and boots up.

Maju3, 2018-05-12:
A optocoupler works different.
It makes a FALLING edge when ignition is ON.
But what about this idea:
Just use your optocoupler and solder it to the PI "RUN" Pins.
This way you just use the ignition to BOOT the PI.
To shutdown it you use the integrated shutdown Features when disconnecting the phone;)

t00fatt, 2018-05-12:
Maju, thank you for your input, but I don't think you understand what I am looking for. Boot up is handled by a separate circuit which actually applies power to the pi. After shutting down, power is cut comepletely from my pi with my setup. However, my optocoupler has an NPN transisitor, so when the ignition signal is on (12v), the output of the optocoupler is LOW. So the ignition turning off would be a rising edge, not falling.

htruong, thank you but that isn't working either. I've resorted to just writing a simple script that handles what I need for shutdown. 

I am still trying to reproduce it consistently, but sometimes my set up for controlling the screen brightness with triggerhappy does not work on some boots. So I'm just going to write my own script that handles everything I need. I was just trying to avoid SSH'ing into my pi every time I update CS to make the scripts executable, but that seems to be a nonissue with the startup.py script.

Maju3, 2018-05-12:
Ah okay now i understand.
Its just that i did not know that you use another circuit to boot the pi.
I thought you do boot and shutdown over gpio3
So what you need is when gpio3 is LOW then the pi do nothing BUT when gpio3 is HIGH then it should shutdown

Maju3, 2018-05-12:
I found something;)

Name:   gpio-poweroff
Info:   Drives a GPIO high or low on poweroff (including halt). Enabling this
        overlay will prevent the ability to boot by driving GPIO3 low.
Load:   dtoverlay=gpio-poweroff,<param>=<val>
Params: gpiopin                 GPIO for signalling (default 26)

        active_low              Set if the power control device requires a
                                high->low transition to trigger a power-down.
                                Note that this will require the support of a
                                custom dt-blob.bin to prevent a power-down
                                during the boot process, and that a reboot
                                will also cause the pin to go low.

Name:   gpio-shutdown
Info:   Initiates a shutdown when GPIO pin changes. The given GPIO pin
        is configured as an input key that generates KEY_POWER events.
        This event is handled by systemd-logind by initiating a
        shutdown. Systemd versions older than 225 need an udev rule
        enable listening to the input device:

                ACTION!="REMOVE", SUBSYSTEM=="input", KERNEL=="event*", \
                        SUBSYSTEMS=="platform", DRIVERS=="gpio-keys", \
                        ATTRS{keys}=="116", TAG+="power-switch"

        This overlay only handles shutdown. After shutdown, the system
        can be powered up again by driving GPIO3 low. The default
        configuration uses GPIO3 with a pullup, so if you connect a
        button between GPIO3 and GND (pin 5 and 6 on the 40-pin header),
        you get a shutdown and power-up button.
Load:   dtoverlay=gpio-shutdown,<param>=<val>
Params: gpio_pin                GPIO pin to trigger on (default 3)

        active_low              When this is 1 (active low), a falling
                                edge generates a key down event and a
                                rising edge generates a key up event.
                                When this is 0 (active high), this is
                                reversed. The default is 1 (active low).

        gpio_pull               Desired pull-up/down state (off, down, up)
                                Default is "up".

                                Note that the default pin (GPIO3) has an
                                external pullup.

With dtoverlay=gpio-shutdown you can select if you want a rising edge OR a falling edge to trigger shutdown on gpio3.

For shutdown on rising edge try this:

dtoverlay=gpio-shutdown,<active low>=<0>
This should trigger a shutdown on rising edge.

htruong, 2018-05-15:
@Maju3 Thank you. Problem solved. Closing the issue :)

closed 2018-05-15 by htruong