opencardev/crankshaft #100: GPS doesn't work when wearable is paired to phone

Archer1632, 2018-05-12:
- What phone do you have? What OS version?
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
- Please provide any further information that you might find helpful.

Archer1632, 2018-05-12:
Galaxy note 8
Android 8.0.0
Pi3 B+  version 2018-04-18
Maps is persistent offline on the Pi when a Galaxy Gear Frontier wifi model is paired to phone.  Able to search for locations and directions download but map will not update while driving and says offline.  

Powered down watch and maps went online.  

Working theory is that the phone does not want to share GPS with two devices and prioritized the watch.

Messages also fail to populate on Pi when watch is on.  Turn watch off or disconnect from phone and messages immediately show on Pi.

htruong, 2018-05-15:
Thanks for reporting the problem. This should not happen. I will see if there is any way I can debug this, but please be patient as I do not have such a device to test.

Archer1632, 2018-05-25:
Just loaded and tested Build 2018-18-05 and it works fine now.  Messages pass to both the device and my watch.  Data flows fine to device when watch is paired.  
I cannot speak for others but as far as what I originally reported this bug is closed.  Thank you for your work!

Archer1632, 2018-05-25:
Sorry... Accidentally clicked close issue.  I will leave that for you to decide if it should be closed or not @htruong

closed 2021-11-27 by github-actions[bot]