opencardev/crankshaft #103: Screen brightness not responding to buttons

Elmardus, 2018-05-14:
I've got two buttons for volume up/down, and two for screen brightness up/down. When pressed, they connect a GPIO pin to ground. The volume buttons work fine, but the (official pi) screen doesn't respond to the brightness buttons.

It sometimes works after a reboot, but most of the time it does not respond to the brightness buttons at all. My gpi2kbd.cfg is set as follows:

```
VOLUMEDOWN      13
VOLUMEUP        19
BRIGHTNESSDOWN  20
BRIGHTNESSUP    21
```

Changing the brightness manually on the homescreen does work correctly. I've also measured the GPIO pins with a multimeter, and it changes from 3.3V to 0V when a button is pressed.

htruong, 2018-05-15:
That is not good - Can you confirm that you are able to change the brightness and could see the brightness button on the left of the OpenAuto main interface?

htruong, 2018-05-15:
I see. Can you move the GPIO pins 20 and 21? Crankshaft at least uses 21 to figure out whether the user wants to rotate the screen in the current version, so it might have conflicted. The default settings should work. https://github.com/opencardev/crankshaft/wiki/Customizing-Crankshaft

I understand using GPIO to signal dev mode/screen rotate/X11 is a nuisance. We'll soon get rid of that GPIO convention and you'll be able to use whatever pin that is convenient to you.

t00fatt, 2018-05-15:
I experienced a similar issue, but was just using GPIO5 and modified triggerhappy to switch between two brightness settings when GPIO5 was pressed or released. It would always work on the first boot, then randomly stop working after one or two boots. I don't recall if it came back on it own or was only after removing the SD card and tweaking gpio2kbd.cfg or triggerhappy.conf. But I did getting working again without touching the hardware setup. I started to fear I had a hardware problem with my GPIOs, so I wrote my own script using the same pin and it has been working perfectly, without touching my hardware set up.

htruong, 2018-05-15:
Sometimes I have experienced a screen rotation pin thinking that it is shorted to ground too. So I thought I was going crazy. I think for some reason the internal pull up doesn't pull up.

If someone can investigate why so it'd be great!

Elmardus, 2018-05-16:
I've changed back to the default pins in gpi2kbd.cfg, however the screen still does not respond to the brightness buttons.

Any ideas what I can try?

closed 2021-11-27 by github-actions[bot]