opencardev/crankshaft #104: Whitelist ofono service

snailium, 2018-05-16:
I'm using the v0.2.3 2018-04-18.

I'm trying to bridge phone call through Raspberry Pi as well. So I installed ofono, and also build up an auto-connect script.

My first try is to auto-connect after bluetooth is ready. It connected to my phone with no problem. While making phone call in Android Auto, it is even showing I'm paired with bluetooth, but no sound through bluetooth.

So I updated my script to auto-connect after both 'bluetoothd' and 'ofono' services are ready. And it becomes never auto-connected. It seems the ofono service is somehow blacklisted and is never started in Android Auto mode.

How can I whitelist the service in Android Auto mode? Could ofono be whitelisted by default?

BTW: just for clarification, I'm using bluetooth dongle for phone connection, as onboard bluetooth doesn't support HFP.

TimBo93, 2018-05-25:
I installed ofono regarding this instruction set as well:
https://scribles.net/enabling-hands-free-profile-on-raspberry-pi-raspbian-stretch-by-using-pulseaudio/

My phone (S8) connects successfully with the device and i can call numbers through python.
But the speakers and mic are not tunneled via bt but instead the speakers and the microphone of my phone are used. :/

Did you get it to work in develop-mode? Do you have any hints what may be gone wrong? Thx!

TimBo93, 2018-06-07:
I changed the 

> raspi-config

 to autologin the user pi.

Afterwards i used
> systemctl --user enable pulseaudio

to autostart the pulseaudio daemon (not system-wide!)

please note that the home-directory config folder is a symlink to /tmp/ sothat you must be aware of your
> ~/.config/pulseaudio 

folder.

PulseAudio and Ofono now works as expected.

hawkeyexp, 2018-08-17:
Coming CS-NG Pre6 this weekend includes first bt inits also for ofono etc. -> closing cause it's now work in progess...

closed 2018-08-17 by hawkeyexp

ofuangka, 2018-08-23:
@TimBo93 

Trying to get HFP working--can you share your Bluetooth connection scripts?  I'm able to connect to my phone in Dev mode using SSH, but ideally I'd like to connect automatically on startup, and I was having some trouble finding examples.

@hawkeyexp I see in the Wiki it says that Bluetooth with HFP is not available. Is it planned?

hawkeyexp, 2018-08-23:
BT auto connect etc. is builtin starting with pre6.

maskaz, 2018-10-26:
> I changed the
>
>
> to autologin the user pi.
>
> Afterwards i used
>
>
> to autostart the pulseaudio daemon (not system-wide!)
>
> please note that the home-directory config folder is a symlink to /tmp/ sothat [...]

Hi, can you tell exactly how did you run pulseaudio service?
It cannot run with ofono, i mean RPI plays A2DP but HFP only when im run pulseaudio from terminal (it only works for me on developer mode)