opencardev/crankshaft #105: Scripts launched in startup.sh get killed

snailium, 2018-05-16:
I'm using v0.2.3 2018-04-18.

I'm trying to put some bash startup scripts in /boot/crankshaft/startup.sh

```bash
#!/bin/bash

# do whatever you need to do here
# by default this does nothing

/home/pi/script.sh &
```

and in the script.sh, I'm using an infinite loop with "sleep 1". From the log file, it seems the script.sh gets killed when it is executing sleep.

So I extended the execution of /boot/crankshaft/startup.sh by adding an extra sleep.

```bash
#!/bin/bash

# do whatever you need to do here
# by default this does nothing

/home/pi/script.sh &

sleep 30
```

Now I can get my script.sh going. But it still get killed after 30 seconds.

What is the correct way to put the startup script at background, and let it run forever?

htruong, 2018-05-16:
Thanks for reporting the bug, we are looking at it.

hawkeyexp, 2018-05-17:
@snailium  - can you please try following:
add "RemainAfterExit=yes" to /etc/systemd/system/user_startup.service after ExecStart in section [Service] and let me know if it works.

snailium, 2018-05-18:
@hawkeyexp Yes, "RemainAfterExit=yes" fixes the problem. I didn't even realize it is a systemd service.