opencardev/crankshaft #106: Screen shuts off every 10 minutes

Maju3, 2018-05-17:
- What phone do you have? What OS version?
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
- Please provide any further information that you might find helpful.

I am using the official raspberry touchscreen.
When my phone is connected and android auto plays music then after 10 minutes the screen goes black.
When i touch on the screen then it shows android auto again.
Is there any screen saver setting activated?

Maju3, 2018-05-17:
Oh crankshaft version is 0.2.2

htruong, 2018-05-17:
I can confirm this as a bug. This will be resolved in future versions. The reason was that the screensaver blanking was not disabled by default.

Ref: https://unix.stackexchange.com/questions/38136/avoid-display-blanking-under-x

Maju3, 2018-05-17:
Okay thank you!
Can you just tell me how i can disable the screensaver?

Maju3, 2018-05-17:
Wait i have another idea.
Maybe some people wants to have a screensaver.
So what do you think about a general setting for screensaver ON/OFF in crankshaft setting screen?

htruong, 2018-05-17:
Please try something like this

```
vi /etc/X11/xorg.conf.d/10-monitor.conf
```
Type in:

```
Section "Monitor"
    Identifier "LVDS0"
    Option "DPMS" "false"
EndSection

Section "ServerLayout"
    Identifier "ServerLayout0"
    Option "StandbyTime" "0"
    Option "SuspendTime" "0"
    Option "OffTime"     "0"
    Option "BlankTime"   "0"
EndSection
```

Maju3, 2018-05-17:
I try it on weekend.
Thanks again!

hawkeyexp, 2018-07-09:
Should be solved in CS-NG correct ? Than issue can be closed.

Maju3, 2018-07-09:
yes,solved in csng pre2

closed 2018-07-09 by Maju3