opencardev/crankshaft #107: Delayed GPIO shutdown

nygma2004, 2018-05-18:
I have a car which does not have a switched 12V line (for the car stereo). The only switched output I have is the cigarette lighter. So I am using that operate a relay and use GPIO3 to shut down the PI.
The problem is that when I start the engine the cigarette lighter connection looses power for a few seconds (probably whilst the starter motor is running).
So I put my car to ignition, radio comes on, PI starts booting. When I start up the engine the radio mutes for a few second. If crankshaft is still booting up it will not monitor GPIO3, and nothing happens, but it happens later, it automatically shuts down the PI.
I hope this is not a unique issue with my car.

So I was wondering if it would make sense to add a delay besides the dtoverlay=gpio-shutdown option. So it only start shutting down if the GPIO3 is LOW for more than 3-5 seconds.

Maju3, 2018-05-18:
Yeah,
the Problem here is that when you start the Motor of your car then the relays looses power for a few seconds and sets gpio3 to LOW and the PI gets the shutdown Signal.
I do not know if it is possible to include a time delay into the "dtoverlay=gpio-shutdown"option.
BUT you could add a simple "off delay" circuit to your existing relays.This way when you start the Motor then the relay will not shutdown.
Something like this:
https://www.amazon.com/Timer-Delay-Relay-Hours-Cycling/dp/B00PD65UGA

Maju3, 2018-05-19:
I have just reworked something;)
https://github.com/opencardev/crankshaft/wiki/Boot--Reboot-and-Shutdown-the-Pi-with-ignition-key

nygma2004, 2018-05-22:
Wow, thanks man. Give me some time to implement and test it.

nygma2004, 2018-05-22:
@Maju3 let me just ask this: you are using GPIO27 because that is something you have used in the past, right? Since I have wired GPIO3 for the built in shutdown script, I can use that as well. I just need to re-wire the relay as the build in script used active HIGH. That is all, right?

Maju3, 2018-05-22:
I do not use this at all.
I use a m3atx psu.
But yes you can use gpio3 for shutdown with this script.
Just edit the input pin in the script to gpio3.
No need to change it to "active high".

Maju3, 2018-05-22:
Nope.
The script i have provided uses active LOW.
Gpio 27 bridged to ground.

To BOOT the pi you need to bridge gpio3 to ground.
Boot works by default.
No script or anything is needed for BOOT.
Just for SHUTDOWN you need the provided script.

BUT you can try to edit the script.
You can edit it to use gpio3 for SHUTDOWN as input.
So the pi monitors gpio3 and when gpio3 goes HIGH then the shutdown should happen.

Maju3, 2018-05-22:
I like your idea but i am not good at scripting at all.
But the idea is really good.
So if you can edit the shutdown script that it does a shutdown when gpio3 is ACTIVE HIGH then you can rewrite my post in WIKI;)

Maju3, 2018-05-22:
Oh and keep in mind that for BOOT you must use gpio3 bridged to ground.
NO other gpio pin can wake the pi from halt state.

Maju3, 2018-05-23:
The dtoverlay=gpio-shutdown option does not work correctly with a relay.
The problem is when you start your cars motor then the relay looses power.
This sets gpio3 to HIGH for a few seconds and would initiate a shutdown.

BUT....
I have a new idea how to use gpio3 ONLY for boot and delayed shutdown with an other script i have found.
Give me a day and i will provide a script here what should do what you want;)

nygma2004, 2018-05-23:
No need to hurry, I am fully loaded with other work and would not have time to test in the next few days.
To be honest, PI works with GPIO3 unconnected as well. I mean once the PI receives power, it starts booting. So that was not an issue for me. 
But I realized that I need to add volume and brightness buttons which means I need to re-wire my GPIOs anyway.

Maju3, 2018-05-23:
[Pi relaysshutdown.zip](https://github.com/opencardev/crankshaft/files/2030485/Pi.relaysshutdown.zip)

Try the script when you find some time.
And tell me if it works for you,please:)

hawkeyexp, 2018-08-02:
GPIO triggered shutdown via ignition will be released with CS-NG Pre4

hawkeyexp, 2018-08-03:
Released - pre4 is on the way.

closed 2018-08-03 by hawkeyexp