opencardev/crankshaft #109: [SURVEY] Which display's you are using ?

hawkeyexp, 2018-05-19:
Hi @ all, I'm working on something new which will come up in near future.

For a special thing i need some feedback so please let me know which display
you are using with crankshaft (only touchscreens which needs no drivers!)
- Manufacturer
- Type/Model
- Settings needed in config.txt
- Output of 'lsusb' for touch controller (usb device id and vendor id)

Sample:

- Waveshare Inc.
- 7inch HDMI LCD (B) Rev 2.1
- Config.txt
   max_usb_current=1
   hdmi_group=2
   hdmi_mode=87
   hdmi_cvt=800 480 60 6 0 0 0
   hdmi_drive=1
- Output lsusb
  Bus 001 Device 004: ID **0eef:0005** D-WAV Scientific Co., Ltd

Please only comment with the specs - thx !

ketanpatil3, 2018-05-20:
7 inch display with 1024 by 600 with capacitive touch screen.

hawkeyexp, 2018-05-20:
@ketanpatil3 - please complete your post like in the sample

closed 2018-07-09 by hawkeyexp