opencardev/crankshaft #111: Nothing is coming up on my display.

AshishGaur28, 2018-05-27:
I'm not even connecting any phone. I simply flashed the SD card with latest crackshaft but when I turn on the display, it only shows vertical black and white lines. And sometimes it keeps on going on and off continuously.
I tried with earlier version as well but that too doesn't seem to be working.

P.S. I checked, my card is working fine. As i flashed it with regular raspbian and it booted up to desktop.

AshishGaur28, 2018-05-27:
I can see the crankshaft screen through HDMI port but not through display port on raspi. This display port works fine with normal raspbian, as i mentioned earlier.

htruong, 2018-05-27:
Please double-check your ribbon cable. The side that doesn't have the text should look like this:

[image]

Then, check the four additional jumper wires. You'll need to connect the 2 pins: Ground (GND/black) and 5V (red) to the 2 pins labelled GND and 5V of the touchscreen. Do not connect the two others you do not need it (but it doesn't do anything bad if you do so correctly).

Please let me know and give me a picture, so I can help you check.

AshishGaur28, 2018-05-27:
I don't know what was the issue. After i checked with HDMI connection, I started to get the homescreen on the display as well. (I didn't change any connection) But still the homescreen comes up after a few auto-restarts only.

Maju3, 2018-05-27:
Vertical lines on the display are depending on a bad contact.
Check the ribon cable,please.

closed 2018-08-03 by hawkeyexp