opencardev/crankshaft #112: Communication error 2 on android auto

Stevereid61, 2018-05-27:
Thank you for all the work on crankshaft. I am attempting to install in a 1927 ford.
I have a issue though and assume it is Android auto version problem as it gi es me a "communication error 2"

Android version 8,android auto version 3.2.581534. Huawei P10

Stevereid61, 2018-05-27:
Pi 3+ and 7" screen. Had to edit line line 25 to force the HDMI to work in the config.txt

htruong, 2018-05-27:
I remember seeing this somewhere. I think this might have been Huawei's fault and there is nothing Crankshaft can do:

https://forum.xda-developers.com/android-auto/android-auto-general/android-auto-compatible-huawei-phones-t3744725

https://www.facebook.com/huaweinotcompatiblewithandroidauto/

Stevereid61, 2018-05-27:
I also have a note 5 that I will try. Thanks

turbolargoo, 2018-06-23:
I am having the same problem on a xiomi mi5 android 7.0 NRD90M miu 9.5.3.0
"Communication error 2 - the phone and the car are running incompatible software"
This same phone and android auto app work on a renault megan, any ideas?
Thanks for all this work

matt2005, 2020-10-17:
Resolved in alpha 55

closed 2020-10-17 by matt2005