opencardev/crankshaft #113: No multitouch with Offical Screen

smartroad, 2018-05-27:
- What phone do you have? What OS version?
Nokia 8/Samsung A5 Oreo

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Pinch to zoom not working in Google maps or Waze

When I try to use pinch to zoom in above nav apps nothing happens just moves with first finger touch. Is there a setting I am missing?

tehniemer, 2018-05-28:
AndroidAuto does not support multi-touch.

tehniemer, 2018-05-29:
Dang, well I stand corrected. I know at one time, not too long ago, it did not.

htruong, 2018-06-07:
This is a known issue. It is possible to add multi-touch, it's just not in the code. Will save for (many) enhancement ideas later :)

borconi, 2019-07-05:
Here is my Java code on how I do multi-touch in Android Auto, you will need to port this to C++ I guess.

InputClass:
```
public void sendInput(TouchEventData.TouchEvent.Builder touch)
    {
        sendInput(touch,null,null,null);
    }
    public void sendInput(ButtonEventsData.ButtonEvents.Builder button)
    {
        sendInput(null,button,null,null);
    }
    public void sendInput(RelativeInputEventsData.RelativeInputEvents.Builder relative, Long evenTime)
    {
        sendInput(null,null,relative,evenTime);
    }
    public void sendInput(ButtonEventsData.ButtonEvents.Builder button, Long evenTime)
    {
        sendInput(null,button,null,evenTime);
    }

    public void sendInput(
            TouchEventData.TouchEvent.Builder touch,
            ButtonEventsData.ButtonEvents.Builder button,
            RelativeInputEventsData.RelativeInputEvents.Builder relative,
            Long eventTime
    ) {
        InputEventIndicationMessage.InputEventIndication.Builder inputData = InputEventIndicationMessage.InputEventIndication.newBuilder();

        if (eventTime==null)
            inputData.setTimestamp(SystemClock.uptimeMillis());
        else
            inputData.setTimestamp(eventTime);

        if (touch!=null)
            inputData.setTouchEvent(touch.build());
        if (button!=null)
            inputData.setButtonEvent(button.build());
        if (relative!=null)
            inputData.setRelativeInputEvent(relative.build());

        InputEventIndicationMessage.InputEventIndication zzz = inputData.build();
        final ByteBuffer sendData=ByteBuffer.allocate(zzz.getSerializedSize()+2);
        sendData.putShort((short) InputChannelMessageIdsEnum.InputChannelMessage.Enum.INPUT_EVENT_INDICATION_VALUE);
        sendData.put(zzz.toByteArray());
        try {
            keySendQue.submit(new Runnable() {
                @Override
                public void run() {
                    try {
                         mTransporter.encrypt(sendData.array(),INPUTCHANNEL,FLAG_ALL);
                    } catch (Exception e) {
                        e.printStackTrace();
                    }
                }
            });
        } catch (Exception e) {
            e.printStackTrace();
        }
    }
    public void sendkey(int keycode, boolean action)
    {
        sendKey(keycode,action,null);
    }

    public void sendKey(int keycode, boolean action, Long eventTime)
    {
        Log.d("HU-KEY","Code: "+ keycode + "action: " + action +", time: " + eventTime);
        sendInput(ButtonEventsData.ButtonEvents.newBuilder().addButtonEvents(
                ButtonEventData.ButtonEvent.newBuilder()
                        .setScanCode(keycode)
                        .setIsPressed(action)
                        .setLongPress(false)
                        .setMeta(0)
                ),
                eventTime
        );
    }

    public void sendScrollKey(int delta, Long eventTime)
    {
        sendInput(RelativeInputEventsData.RelativeInputEvents.newBuilder().addRelativeInputEvents(
                RelativeInputEventData.RelativeInputEvent.newBuilder()
                        .setScanCode(65536)
                        .setDelta(delta)
                ),
                eventTime
        );
    }
```

And the touch listener:
```
@Override
    public boolean onTouch(View v, MotionEvent event) {

        if (mContext.mService==null) //Thigh coupling this should never happen!
            return false;

        event.getActionIndex();
        int me_action = event.getActionMasked();

        Log.d("HU", "Event pointer counter: " + event.getPointerCount());

        TouchEventData.TouchEvent.Builder touchdata = TouchEventData.TouchEvent.newBuilder();

     for (int i=0;i< event.getPointerCount(); i++ ) {
         Log.d("Touch","View height: " + v.getHeight() + ", display Height:" + BuildCarConfig.screenHeight+", Vertical adjust: "+verticalAdjust + "Touch pos: " + event.getY(i) + " Converted: " + ((int) ((event.getY(i)-verticalAdjust) / (v.getHeight() / (float) m_virt_vid_hei))));
         touchdata.addTouchLocation( TouchLocationData.TouchLocation.newBuilder().setX((int) ((event.getX(i)-horizontalAdjust) / (v.getWidth() / (float) m_virt_vid_wid))).setY((int) ((event.getY(i)-verticalAdjust) / (v.getHeight() / (float) m_virt_vid_hei))).setPointerId(event.getPointerId(i)).build());
         //touchdata.addPointerData( Input.TouchEvent.Pointer.newBuilder().setX((int) (event.getX(i) / (v.getWidth() / (float) m_virt_vid_wid))).setY((int) (event.getY(i) / (v.getHeight() / (float) m_virt_vid_hei))).setPointerId(event.getPointerId(i)).build());

     }
        TouchActionEnum.TouchAction.Enum aa_action;
        switch (me_action) {
            case MotionEvent.ACTION_DOWN:
                aa_action = TouchActionEnum.TouchAction.Enum.PRESS;
                break;
            case MotionEvent.ACTION_MOVE:
                aa_action = TouchActionEnum.TouchAction.Enum.DRAG;
                break;
            case MotionEvent.ACTION_CANCEL:
                aa_action = TouchActionEnum.TouchAction.Enum.RELEASE;
                break;
            case MotionEvent.ACTION_UP:
                aa_action = TouchActionEnum.TouchAction.Enum.RELEASE;
                break;
            case MotionEvent.ACTION_POINTER_DOWN:
                aa_action = TouchActionEnum.TouchAction.Enum.POINTER_DOWN;
                break;
          case MotionEvent.ACTION_POINTER_UP:
                aa_action = TouchActionEnum.TouchAction.Enum.POINTER_UP;
                break;
            default:
               return false;
        }

        touchdata.setTouchAction(aa_action);
        touchdata.setActionIndex(event.getActionIndex());

        inputMessage.sendInput(touchdata);

        return true;
    }
```