opencardev/crankshaft #116: Audio does not work

tonin233, 2018-06-04:
Hello, the following problem occurs to me. Try to put the raspberry pi with the screen in the car, to test the audio, simply connect an auxiliary cable from the raspberry connector to the auxiliary of the car radio, and this did not work. This should work? Is there any way for the audio to work without having to buy a module?
Thank you.

tehniemer, 2018-06-04:
The 3.5mm jack on the pi includes video and the pinout is different than most. Using a regular aux cord won't work with it.

tonin233, 2018-06-04:
So if I use a cable of the following, should the audio work properly?

Model B+ | Video | Ground | Right | Left |  
Apple | Video | Ground | Right | Left |  
Zune | Video | Ground | Right | Left |

tehniemer, 2018-06-04:
Yes, those will have RCA connectors, but you can convert back to 3.5mm from that pretty easily if you need to.

Maju3, 2018-06-04:
BTW if you use the 3,5mm audio jack from the pi this has very poor audio quality.
To improve the quality do this:
In config.txt add this:
audio_pwm_mode=2

tehniemer, 2018-06-04:
One thing I noticed using the built in 3.5mm port is that the volume was inconsistent and pretty low. Nothing I tried fixed it for me so I ended up getting a HiFiBerry.

tonin233, 2018-06-05:
vale gracias lo probare, pero si utilizo un jack to USB en la raspberry, esto debería funcionar?
[image]

htruong, 2018-06-07:
@tonin233 looks like your first picture is an iPod charger cable, and it will defnitely not work, and don't use it, it might fry your equipments.

The second one should work, but I do not have it on my hands to test.

chunky74, 2018-06-07:
The USB cable looks like the Apple iPod Shuffle - That won't work.
I can confirm that the image of the USB 'Soundcard' does work!  I have one in my car working right now.  I did have trouble with the Mic.  I am using a two connector (mono) Mic, the soundcard is wanting a 2 channel Stereo Mic(3 connectors).  I did get the mic to work by wiggling/pulling it out slightly.

tonin233, 2018-06-08:
@htruong @chunky74 
Thank you very much to both of you, I will try it and comment if it works.

DJFliX, 2019-01-19:
Hi @tonin233! Can you retest and report if you are still experiencing the issue?

matt2005, 2020-10-17:
Stale issue

closed 2020-10-17 by matt2005