opencardev/crankshaft #117: Waze App Screen going black.

hahmed149, 2018-06-04:
- What phone do you have? What OS version?
One Plus 3T  | Oxygen OS 5.0.3 | Android 8.0.0
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
The bug is produced when I launch Waze app on the touch screen connected to the Rraspberry Pi, in rare occasions, it launches but after a minute or two the screen goes black with the home button bar on the Pi's Screen still showing. 
- Please provide any further information that you might find helpful.
Here's exactly what I did. 
1. Downloaded the latest release of crankshaft and using etcher got it on my SD card. 
2. Installed android auto app on my phone ran through the initial setup. 
3. Enabled Developer mode
4. Plugged in my phone to the Pi.
5. Launched Waze using Raspberry Pi's Touch screen. (I am using the office 7" touchscreen display)
6. 50% of the time the screen goes blank with the home bar still appearing on the 7" touch screen however the system becomes unresponsive. 

Please let me know, if you want any specific information from me or if I can contribute to resolving this issue in any way.

vistazifta, 2018-06-11:
Try disabling "developer mode" on your phone. It is known to cause issues.

hahmed149, 2018-06-11:
@vistazifta Does the "Developer Mode" has to be turned off on the Android Auto App, or on the Developer Options on the phone itself?

t00fatt, 2018-06-17:
I've found that I have to have waze already open and in the background before connecting my phone to get waze working in crankshaft. Also, if I dont have spotify or pandora already open on my phone, I get an error along the lines of "spotify/pandora doesn't seem to be working right now."

hahmed149, 2018-06-17:
@t00fatt I tried having waze open and then connecting it to android auto, it seems to work fine like this. Not sure if its a crankshaft bug or an Android Auto issue with waze.

hawkeyexp, 2018-07-09:
Still present in CS-NG ? If not please close issue.

hahmed149, 2018-07-10:
I will test it tonight and close this issue if it is resolved.

hawkeyexp, 2018-07-10:
Ok let me know or close the issue yourselve if fixed in cs-ng.

hahmed149, 2018-07-11:
The issue is still not resolved.

hawkeyexp, 2018-08-03:
I have checked it with latest waze and it seems running without problems over 1h - Possible a power management issue etc. on device side ?
I think it is not on CS side so closing.

closed 2018-08-03 by hawkeyexp