opencardev/crankshaft #118: 10.1" Touchscreen Not Working

iandunham2, 2018-06-06:
I'm trying to use Crankshaft using this touchscreen:

https://www.amazon.com/gp/product/B01H04GENA/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

I can't get any touch response at all. I've run C.S. in Dev Mode, and have installed the eGalax drivers and have tried following this tutorial: https://github.com/opencardev/crankshaft/issues/64

But I can't get any response from the touchscreen.

C.S. definitely sees the touchscreen controller, when I run lsusb, it shows up as a USB object. There is just no response out of it.

Also, I tried installing Raspbian, and it recognizes the touchscreen right away.

htruong, 2018-06-07:
Did you start OpenAuto in X11?

https://github.com/opencardev/crankshaft/wiki/Custom-touchscreens

iandunham2, 2018-06-07:
Yes, I tried that. The screen and touch layer behave the same.

closed 2021-11-27 by github-actions[bot]