opencardev/crankshaft #123: Feature Request: Enable Location from Vehicle Screen

Aaron407s, 2018-06-11:
I'm currently running a Samsung Galaxy S7, Android 7.0 stock with Crankshaft. After some tweaking it seems to work fairly well overall. However, one things that's a bit of an annoyance is trying to use navigation if location isn't enabled on the phone as it forces you to unlock the phone to enable it. Is it technically feasible to have a dialog with buttons on the vehicle screen to allow you to enable location, even if it's at the time of initial connection? I realize that location could be automatically enabled on AA launch if the phone's rooted, but I'm curious if there's another option through the pi's interface without rooting.

Maju3, 2018-06-13:
Are you sure that location is not enabled?
I just ask cause i have a asus zenfone with android 7.1
My phone is NOT rooted but when i start android auto then location is enabled by default.

Aaron407s, 2018-06-16:
I did an update of my phone, and it seems like it may be inconsistent now. I'll report back after trying a bit more, but need to get my AR1100 touchscreen overlay working to properly test things...

Maju3, 2018-06-16:
Okay i see.
But why do you not just turn location ON on your PHONE?
I mean BEFORE you connect your phone to the pi.

Aaron407s, 2018-06-17:
That's certainly an option, but I was hoping that it could automatically enable or be prompted on the vehicle screen to enable it, especially since AA launches automatically as soon as the phone is connected so you don't have to manually open it. I prefer to normally leave location off to save on battery. It just seems odd that you'd have to pull out your phone in the middle of a drive when you want to use navigation, just to turn location on if you forgot to do it originally.

Maju3, 2018-06-17:
Yes i see what you mean.
But this is not related on crankshaft.
When you turn off location manually then android auto is not able to turn it on.
It might work on rooted phones but i don't know.
For me in person i just leave on location all the time or just turn it on before i enter my car.

Maju3, 2018-06-17:
Just one thing about location setting on your PHONE.
This setting is not really to turn location on/off but it is to ENABLE/DISABLE it.

That just means you can ENABLE location all the time and as long no app uses location it is NOT turned on and uses no battery.

Just try this:
Enable location on your phone settings.
Start Android auto on your PHONE.
On your phones status bar you will see that android auto turns on location.
When you exit android auto you will notice on the phones status bar that location turns off after exit.

So you can ENABLE location on the phone all the time.(like i do it).

Aaron407s, 2018-06-18:
Thanks for the clarification. I noticed this a few times but wasn't sure exactly how it worked. I always kept location disabled after having battery drain issues with past phones, but OS handling of location has probably improved since then.

Note, however, that I'm not necessarily looking for it to automatically enable location. In the event that location isn't enabled, it would be great to even have a button on the screen from my previously attached photo where you can simply select "enable location" rather than having to enable it on the phone. I don't know the ins and outs of AA or Crankshaft, though, so I'm not sure if standard permissions would allow this.

Maju3, 2018-06-18:
Nope.
Standard permissions do not allow this.
Your phone need to be rooted for this.

You can try this very simple:
DISABLE location in your phone settings.
Start AA app on your PHONE.
Make sure that AA has permissions for location.
You will see that AA app can NOT enable location on your phone.
It might work only on ROOTED phones.