opencardev/crankshaft #124: Feature Request: Music Auto Play on Phone Connect

t00fatt, 2018-06-17:
Would it be possible emulate a play/pause button press after a phone is connected so music will automatically start playing?

Is there anyway to check the phone connection state in a separate script without modifying the crankshaft build itself?

nthanhtin, 2018-06-20:
Isn't it already implemented in Android?. In my case it automatically plays music when I plug my phone in and stops when I pull it out.

Maju3, 2018-06-20:
Really?
In my case i need to click on the play button to start music playback.

htruong, 2018-06-23:
Some player program such as Rocket Player (below) has the feature to automatically play upon connection. I think you can try out that feature.

[image]

hawkeyexp, 2018-08-09:
A feature like auto play on connect could be nice but it should not be done by cs. There are solutions like players which can handle it or for example automation tools like tasker which can do the job for advanced users.

closed 2018-08-09 by hawkeyexp