opencardev/crankshaft #126: Proper Support for AR1100 Touch Screen Controller

Aaron407s, 2018-06-21:
I'm using a resistive touch screen with an Adafruit AR1100 controller, which outputs as a USB mouse, Generic HID device, or digitizer depending on the selected communication mode. Everything is running on a Pi 3B, although the issues below apply to a 3 B+ that I also tried. Although one guy with what seems like the exact same setup claims to have it working, I can't get it to work properly for touch control. In both X11 and EGL mode, I can get it to register touches on the screen only as mouse clicks if I have an additional mouse connected to pick the position. In X11 with the controller set to digitizer mode, I can get it to act as a mouse where I can drag the pointer across the screen with continual finger swipes, but it doesn't register the touch coordinates in any mode. I've tried disabling the 'enable touch screen' option, but that didn't help as I presume that's for enabling a touch screen via GPIO.

The controller was calibrated through windows, and works fine with windows 7 where it's recognized as a generic usb mouse and the mouse pointer jumps to the point where the screen is touched. It also works exactly as it should in Raspbian Jessie. It just doesn't work in Crankshaft.

Is there any hope? Everything I've installed is basically sitting in limbo...

tehniemer, 2018-06-21:
I had a very similar issue when I installed one. Mine was damaged but displayed the same symptoms of working fine on Windows but not on Linux. I bought a new one and all is well. Be very careful not to damage the USB connector it is a weak point. Have you inspected the board? Or have you tried different USB cables?

tehniemer, 2018-06-21:
I would be willing to bet that it's the HDMI converter, I had zero luck getting those to work for this, and I tried a few. For shits and giggles, try setting your overscan compensation to 0 across the board in ```config.txt``` and see if that does anything.

Aaron407s, 2018-06-21:
What's strange is that it was outputting the full frame without any issues before the touch screen controller was attached. I'll give the overscan compensation a try, though. Strangely, it outputs the full video frame after connecting my phone and entering open auto, but the touch screen can't control outside of that cropped block.

tehniemer, 2018-06-21:
Huh, I can't really help much with the HDMI converter, but Zune cables are ~$7 on amazon with prime, and you won't lose any image quality with that screen.

Aaron407s, 2018-06-21:
It was more that I could still access the menus in the main screen by using 720p. I may have to go with the cables, though.

I am narrowing down the source of the problem, though. It seems that running it under X11 is what causes the strange issues. If I have a freshly flashed card with nothing changed, it outputs properly to the screen but the touch screen doesn't work (except for simple clicking). If I take that same card and enable X11, it does the strange cropping and the touch screen will only work in that area. There almost seems to be a conflict between the screen resolution and touch screen area when using HDMI. I have no idea how to fix that, but I may buy a Zune cable and see if it works any differently, even if it does cause the scaling issue on the main screen.

Aaron407s, 2018-06-21:
I tried putting overscan to 0 across the board, but it didn't help. It's odd how just a simple change from EGL to X11 causes the screen reduction/cropping issue and the touch screen doesn't expand beyond that size. I'm going to try forcing some smaller resolutions to see if it changes anything, and I may as well track down a video cable. Unfortunately, the cables don't seem to be as quickly available in Canada...

Aaron407s, 2018-06-21:
Ok, getting closer. I can force a lower HDMI output resolution for it to not crop and expand the touchscreen area. It should just be a matter of tweaking it from there, ultimately to get the same result as using the analog cable in terms of the menu spilling outside of the screen. However, I can tweak that stuff on a different display if necessary. I'm happy to see some progress at any rate, and I could actually push each of the buttons in open auto with the reduced resolution.

tehniemer, 2018-06-21:
The native resolution of that screen is 400x234 if that will help.

Aaron407s, 2018-06-21:
Since it's running through the HDMI converter, it probably doesn't care about the native display resolution on the end of the RCAs, but I've been wondering what the resolution is anyway. Right now, a 480p HDMI output gives probably 80% screen coverage. I'm going to play around with over/underscan and see if I can get it to fill properly. Definite progress at any rate.

Aaron407s, 2018-06-21:
I got it working.

For anyone else fighting with the same issue, I had to use HDMI group 1 mode 3 and a lot of trial and error for overscan (currently set at L -32, R -28, T -30, B -30). I replaced the wallpaper with a different 800x480, but it only shows the left 720x480, likely due to the mode 3 resolution setting. I could probably just use a 720x480 png, but I won't worry about it for now.

The touch screen works well now across the full screen. The only issue I have is that I can't increase the screen DPI beyond the stock 140 to get larger buttons. If I do, I get an incompatibility error in AA on my phone. I was hoping to use 160 as I had previously found it to be a nice size for this screen, but I can live with it as is.

Aaron407s, 2018-06-21:
Re-opened for the screen DPI limitation as noted.

closed 2018-06-22 by Aaron407s