opencardev/crankshaft #127: Google Assistant Recording "What's the message" when responding orally to WhatsApp messages

emstone1, 2018-06-21:
I have my phone sending Bluetooth audio directly to the car system. This approach appears to be the recommended route in the documentation as the audio quality, especially for music, sounds terrible when routed through the PI. I have a USB microphone connected to my PI to capture my voice. When I try to reply to a WhatsApp or SMS message, I believe the delay caused by the Bluetooth is causing the system to record "What's the message" prior to my speech. This is what the google assistant is saying. It appears, as I said, that this may be due to the delay caused by using Bluetooth to transmit this message but I am not sure. I am wondering if we couldn't either fix this bug completely or add a setting to create a delay before the microphone activates?

- What phone do you have? What OS version? 
**Pixel, 8.1.0**
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always reproducible.
**Every time I try to reply to a message through Android Auto.**
- Please provide any further information that you might find helpful.
N/A

htruong, 2018-06-21:
This seems to be a bug in the Android Auto app. I am also experiencing it. I read it a couple of weeks ago someone is experiencing it on the real head units. 

I will keep this open to monitor the issue.

chunky74, 2018-06-21:
May not be a fix - But cant you setup Voice control on your phone to ONLY respond to your voice ?  Short term that would reject the voice saying Whats the message ???  Cant find the setting to be honest :-(  But could have sworn there was one.

Aaron407s, 2018-07-18:
I find this to still be an issue even without bluetooth. AA ends up recording part of its own voice, especially when dictating text messages (e.g., When asked "What is your message?" and I respond with "Be there soon", the message will be read back as "Message be there soon". If you check the Android Auto User Community for the post "Android auto records itself", you'll see that it's an issue dating back about a year and a half that, even with about 500 posts in the thread, doesn't seem to be getting fixed.

closed 2021-11-27 by github-actions[bot]