opencardev/crankshaft #128: Bypass Lockouts?

Aaron407s, 2018-06-22:
This may be more a function of AA than Crankshaft, but is it possible to bypass the music browsing lockouts? For example, when moving, trying to browse artists/songs in Rocket Player or any other app that lets you browse will cause a 'locked out for safety' message or something to that effect. I drive a lot of gravel roads where the microphone doesn't pick up voice commands properly, so the ability to browse through the screen is rather critical. I'm guessing that this is caused by a flag in AA rather than Crankshaft, but it would be great to bypass this.

Maju3, 2018-06-22:
Yes i absolutelly agree with you!
Bypassing this browsing limitation would be great.
I know browsing is limited by google cause of "savety".
But bypassing this would allow people on the passenger seat to browse for songs at least ;)

htruong, 2018-06-23:
I think you should make this case with Google :) because we (CS/OpenAuto) have absolutely no control over whether they disable browsing or not. There is an Android Auto Google Groups somewhere...

Aaron407s, 2018-06-23:
I'm admittedly a bit ignorant of where AA ends and CS begins. I was hoping that there might be an ability to make it think that it was stationary to stop the lockouts, but I'm sure that would break navigation if it were possible.

In terms of the AA google group, I was reading it yesterday and many, many people have said that they want the lockouts removed over the last year or so, but the AA developer doesn't seem to have any intention of doing so. It's too bad that there isn't a developer mode that removes them while absolving google of liability.

Maju3, 2018-06-23:
Yes,some kind of hack is needed to remove this stupid lockouts.
And then a Setting like "remove safety Settings" ;)
With the hint: "Use at our own risk" :)
Seriously this browse lockout is very annoying.

borconi, 2019-06-06:
Set car speed to 0 every 400ms and that should do the trick,