opencardev/crankshaft #130: PD injector / Quick Charge Injector?

cydia2020, 2018-06-28:
Can anyone confirm whether [this](https://au.mouser.com/ProductDetail/ON-Semiconductor-Fairchild/FUSB302UCX?qs=sGAEpiMZZMtv%252bwxsgy%2fhiOxaT%2fU6yFuH6iEDFf1qEos%3d) chip will work as a PD injector or not?

This chip seems to be simple enough to work with, and it's fairly cheap.

The charging current from the Pi is painfully slow, my phone only tops up about 3% each hour.

emstone1, 2018-06-28:
I don't have an answer to your question but I did improve my situation of charging buy buying a USB Y Cable which allows for a separate power source to your phone. You connect the dual end to PI (data only) and power and the other end to your phone.

cydia2020, 2018-06-28:
I did think of that, but I don't really want to take the risk of frying my phone as the car charger's output is not so consistent. Using an injector will be a much better thing to do. [This](https://www.amazon.com/Coolgear-Delivery-Injector-Type-C-Device/dp/B075F6BD4Y) probably will also work tho the price is a bit steep.

hawkeyexp, 2018-08-17:
I think should be discussed on reddit not on an issue. -> close.

closed 2018-08-17 by hawkeyexp