opencardev/crankshaft #132: Phone calls and some audio problems

TareqSaleem, 2018-07-01:
Hello,

First of all, thank you for this project and how easy it is to install. I have installed Crankshaft on a Raspberry Pi 3 with the official Raspberry Pi screen. The problems I have are:

1. A peeping noise and crackles and the same time the moment I start the ignition.
2. Google Assistant dose not work after the OK Google phrase (It recognize the OK Google phrase but does nothing afterwards).
3. If I call from Android Auto, the phone uses its speakers instead of the car's speakers.

I used a USB sound card with a headphone jack and a microfoon jack combo.

- What phone do you have? What OS version?
Galaxy S8, Android 8.0
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Every time I use Android Auto
- Please provide any further information that you might find helpful.
N/A

Yantastic, 2018-07-02:
2. Do you have an external Mic attached to the Pi? I'm pretty sure the 'OK Google' phrase is detected using the phone Mic and everything else is offloaded to the headunit (in this case Pi) Mic for detection. I've got one of these and it's working a treat https://www.amazon.co.uk/gp/product/B00IR8R7WQ/ref=oh_aui_detailpage_o05_s00?ie=UTF8&psc=1

3. Similar to 2, the phone call audio is routed through the phone if no external audio devices are present. On the screen you'll probably see 'Bluetooth not connected' or something similar. Again, one of these is working great for me https://www.amazon.co.uk/gp/product/B078W1HQCK/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1

TareqSaleem, 2018-07-13:
Thank you for the reply.

You are probably right about 2 and 3, I'm using an external sound card where I could attach a microphone and an AUX cable, This one to be exact:
https://aliexpress.com/item/Original-Chipset-Antimagnetic-External-USB-Sound-Card-7-1-Adapter-3D-Audio-Headset-Microphone-3-5mm/32584426708.html?spm=a2g0s.9042311.0.0.20184c4dkq1FKk

I think that the sound card is not working properly and not detecting the microphone or the microphone is at fault. I'm going to have to test them separately.

PS: Sorry for the late reply, I had final exams and didn't have the time.

hawkeyexp, 2018-07-16:
Stay tuned - a gui config to select audio device for in/out will be available with pre4. So i think most problems will be easy solved.

TareqSaleem, 2018-07-16:
"Stay tuned - a gui config to select audio device for in/out will be available with pre4. So i think most problems will be easy solved."
Thank you very much, that's great to hear.

hawkeyexp, 2018-08-19:
What's the current state with latest build ? (CS-NG Pre6)

TareqSaleem, 2018-08-19:
The latest build I tried was Pre4 and it didn't resolve my problems, it added one where only the left speakers were working so I reverted to the stable build and now it works.
The noise that I had is now gone by attaching a [ground loop isolator](https://www.aliexpress.com/item/Fornorm-Ground-Loop-Noise-Isolator-Audio-Anti-jamming-Device-Noise-Isolator-with-3-5mm-Audio-Interface/32832643100.html?spm=a2g0s.9042311.0.0.3f144c4dxz6U0h)
The stable build is also faster to start on my raspberry pi 3b so I stuck with it.
I will give Pre6 a try when I have sometime.

Thank you for adding features and keeping it updated, I will let you know if something happens.👍

With kind regards,

Tareq

hawkeyexp, 2018-08-19:
Ok i will close the old issue - feel free to open a new one if needed.

closed 2018-08-19 by hawkeyexp