opencardev/crankshaft #133: [csng] Kodi does not launch, stuck at login screen

dshokouhi, 2018-07-02:
- What phone do you have? What OS version? Pixel 2 XL 8.1 Crankshaft-NG Pre1 2018-07-01
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.

1. Install latest crankshaft-ng
2. Login to dev mode and install kodi via `sudo apt-get install kodi`
3. launch `kodi`
4. Verify `/usr/bin/kodi` exists
5. Verify `/home/pi/.kodi` exists
6. Start crankshaft normally
7. Click on Kodi icon

User is stuck at login screen asking for password.

- Please provide any further information that you might find helpful.

dshokouhi, 2018-07-02:
Also, I realize Kodi is not supported but thought I would mention this since it did not work for me.

Raspberry pi model 3b+ using 5" waveshare screen

hawkeyexp, 2018-07-02:
let me have a look - last tested it was working...
ps: egl or x11 mode if csng ?

dshokouhi, 2018-07-02:
Thanks for the quick reply! I believe egl as I never enabled x11 mode?

All I really did was follow these steps to install my touchscreen: https://waveshare.com/wiki/5inch_HDMI_LCD

hawkeyexp, 2018-07-02:
ok i identified it - kodi tries to create it's home folder on /home/pi which is read only in normal mode so it can't be created. In dev mode you should be able to launch kodi. I will think about this behaviour later.

hawkeyexp, 2018-07-03:
Solution done for pre2

closed 2018-07-06 by hawkeyexp

Richard6360, 2018-07-27:
is there a install script for kodi ?

Richard6360, 2018-07-27:
how to login at dev mode ?

pumpkintechlabs, 2018-07-27:
@Richard6360 no there isn't. You need to get into dev mode and install Kodi as per instructions in the original post (sudo apt-get install kodi). You also need to run crankshaft in X11 mode in order for Kodi to work.

commit 548ef6bf (2018-07-03): [script] Intruduce CSSTORAGE as external rw storage / fix #133