opencardev/crankshaft #135: Noise from CPU

Maju3, 2018-07-03:
Hi.
I use a USB dac connected from PI to my amps.
When I playback music or when the navigation voice comes up then the playback is clear and i have no crackles or other sound issues.
ONLY when I PAUSE the music then I can hear some noise(sounds like cpu noise or access to sd card).

Now my Question is:
Is it possible to make it that pulseaudio play SILENCE as soon as crankshaft starts?
I THINK playing SILENCE COULD solve noise comming over the USB line when NO music plays.

andreacampanella, 2018-07-03:
The noise you hear if from your car, try to change your usb charger

Maju3, 2018-07-03:
I think it is not from my car.
I have used the same m3atx psu snd the same usb dac in my previous car pc install based on a mini itx mainboard.
And there was NO noise on usb.

hawkeyexp, 2018-07-04:
One idea: if you are owner of a power bank lets run the system for a test from this and check for the noices.

andreacampanella, 2018-07-04:
You should try with an audio isolation transformer imho.

Maju3, 2018-07-04:
The powerbank is a good idea:)
i check this at weekend.
The crazy thing is that this noise is NOT permanently present.
Just to better explain it:
When the pi boots there is NO noise.
When i plug in my phone is NO noise.
BUT when AA comes on the screen then the noise starts and as soon as i start music playback the noise goes away.
Even if i pause the music there is NO noise at all.
Next to it when i listen to internet radio and drive through a area with bad internet then the radio app tries to buffer and then the noise comes up again.
Thats why i think it has something to do when the pi needs to work(cpu or something).
BTW some forum says that usb audio from the pi is not the best solution.
A hat like the hifiberry dac or allo boss dac should be much better.

smartroad, 2018-07-08:
As I understand it, the issue is the Pi has very noisy digital power and unless the DAC (USB or otherwise) isolates the analogue supply from the digital you will get noise from the CPU, WIFI, SD card access etc. It is the same as using the internal sound.

Maju3, 2018-07-08:
@smartroad 
Yes you are right.
Today I have powered the pi with a powerbank so the power is completelly isolated from my cars power.
The noise is still present.
What is curious is that the noise comes only when NO Audio (Music or Google assistant) is playing.
Hard to describe but it Sounds a bit like a old hdd you know(clicking,crackling).
But as soon as i start Music the noise is gone.
Even when i PAUSE the Music then the noise is NOT present but after a while it Comes back.

Now my Question:
Is it possible to make a Setting in pulseaudio or anywhere else that the pi Plays a signal of SILENCE when no other Audio(Music,google asistant) Plays?

hawkeyexp, 2018-07-08:
I will have a deeper look on it for later releases. Currently bringing up Pre2.

xstsx, 2018-07-08:
Maju3, from what you described, try "exit-idle-time" option in daemon.conf of pulseaudio. 
From man:
IDLE TIMES
exit-idle-time= Terminate the daemon after the last client quit and this time in seconds passed. Use a negative value to disable this feature. Defaults to 20. The --exit-idle-time command line option takes precedence.
scache-idle-time= Unload autoloaded sample cache entries after being idle for this time in seconds. Defaults to 20. The --scache-idle-time command line option takes precedence.

Set it to -1 , there will be always "silence"

Maju3, 2018-07-08:
@xstsx
Thanks for your suggestion.
But "exit idle time" is already set to "-1".

Does't solve it.

Maju3, 2018-07-08:
Tomorrow i try with " high priotity" and "cpu limit=no" settings

Maju3, 2018-07-09:
Okay,tested with csng pre2.
Pulseaudio daemon settings:
Cpu-limit=no
High-priority=yes
Exit-idle-time=-1
Default-sample-rate=48000
Resample-method=soxr-hq
Default-fragments=8
Default-fragment-size-msec=5

Music playback is perfect.
Navigation voice is good but sometimes it has time outs and delay.
E.g. It should say " drive to north".
Then it says " drive to".
Then music plays.
Two seconds later music pauses and google assistand says "north".

About the noise that comes up sometimes it is still present.
It comes NOT directly when i plug in my phone but it comes up when AA screen is loaded.
As soon as i start music it goes away.
And sometimes when i pause music for a minute it comes again.
Same when i drive through a area with bad internet and the radio app needs to buffer...

Maju3, 2018-07-12:
Now.
A few mins ago i have purchased an Allo Boss Dac.
It should arrive in a few days.
Then i will see if this noice comes from my usb dac/Pi usb bus or it is related on anything else.
We will see;'

hawkeyexp, 2018-07-13:
Problem identified - caused by mobile radio. Better schielding needed.

closed 2018-07-13 by hawkeyexp