opencardev/crankshaft #138: (csng) Freezes when phone disconnect and reconnect

Maju3, 2018-07-03:
I have noticed that when i disconnect and reconnect the phone quickly a few times then CS completelly freezes.
On mainscreen is the message"phone connected" but AA is not shown.
And it does not reakt on clicks at shutdown or settings button.
Only a hard shitdown by pulling power solves this.

Can anyone confirm this?

hawkeyexp, 2018-07-03:
In my case it only had freezes if power supply was to weak - make sure your power supply can deliver enough to load the pi including a connected phone which wan'ts to charge. Possible you can include an active usb hub which can power the phone and the pi only needs it's own power.

Maju3, 2018-07-03:
I use a M3ATX power supply.
6 ampere max at 5 volt line.
With this psu i power the PI AND a powered usb hub.
My phone and my usb dac is connected to the usb hub.

Maju3, 2018-07-04:
I found the problem.
When i connect my phone DIRECTLY to the pi then all works.
BUT when i connect my phone to the USB HUB then after disconnecting the phone CS freezes and on mainscreen ist still the message "phone connected".

hawkeyexp, 2018-07-04:
Ok if it works on direct connect it tells me system has no prob and is fine. The phone connect string is generated if udev detects phone and only shows the physical connect of the phone.
About the freeze i think a cross check with another usb hub would be a good idea.
Generally i wan't to close this issue cause it seems to be caused by 3rd party hardware.

Maju3, 2018-07-04:
Sorry but today i have tested this again.
The phone is connected directly to the pi.
Same issue.
When i disconnect and connect the phone a few times then CS crashes.
This means when AA is shown no button clicks react.
And when i disconnect the phone then the mainscreen comes up and still shows "phone connected".
And even on mainscreen no button reaction.
I have to say that usually i connect my phone BEFORE i boot the pi.
Maybe this causes the problem on csng?

I have another sd card with crankshaft 0.2  and have tested with it.
No problems at all.
I think there is a bug in csng about phone connection on usb.

hawkeyexp, 2018-07-04:
Some clear words:

- why often disconnect and reconnect phone ? usually you connect use it and thats it
- Phone connected state can be going crazy if you connect and disconnect repeatly in short time case it has a delay ~ 1sec to switch state
- Connect before boot is usually possible - in my case no prob
- I think your problems are caused by exernal parts like usb hub / power supply etc. - the pi needs a stable powers supply - any freezes i had with rpi's was caused by psu or to high consumption usb devices connected - also some usb hubs are not really optimal and can cause probs.

Maju3, 2018-07-04:
The problem is that after about 5 minutes using AA no button in AA works.
It seems like AA freezes.
THAT is why i disconnect the phone and reconnect it.
But then even on mainscreen no button works.
Only way to solve it is to restart the pi.

hawkeyexp, 2018-07-05:
Ah ok - now we are on a new and complete different way :-)

We can try to identify what happens with debug mode :-) Debug mode is a new feature primary for internal testing. You can enable it by placing a dummy file called `ENABLE_DEBUG` on a usb stick and connect it before boot or do a reboot after connect. This allows system to startup as normal system with enabled timers read only filesystem etc. but with enabled network and ssh. The interesting part would be if system is still keep alive and let you login and only openauto lost touch connectivity or system is really freezing. Try this out please.

Maju3, 2018-07-05:
Well not only openauto lost touchscreen connectivity but the complete system is freezing.
Because as said even if i unplug the phone a click on any button in MAINSCREEN has no effect.....
And i am pretty sure that is not related to my hardware or powersupply.
Because when i use crankshaft 0.2.2 or as today tested with 0.2.4 EVERYTHING works.
No freezes,all buttons presses works in AA and on mainscreen.
SAME power supply,same usb cables and same usb hub.
Only difference is the software.
BTW there is another post which describes exactly the same issues.

Maju3, 2018-07-05:
Issue #139
User "Hisma" seems to have exactly the same behaviour.
Freezes,touchscreen not working,phone disconnected but message "phone connected" on mainscreen.....

hawkeyexp, 2018-07-06:
Possible a bug is found - have to been tested again if still present in upcoming Pre2

Maju3, 2018-07-06:
Excelent!
I test it on next csng release.
THANK YOU!

Maju3, 2018-07-09:
Solved with csng pre2.
Touchscreen works without problems.
No freezes.
Great work man!

closed 2018-07-09 by Maju3