opencardev/crankshaft #139: Display issues with Crankshaft NG

chrisfromwa, 2018-07-03:
- What phone do you have? What OS version?  Samsung Galaxy S8/Android 8.0
Latest version of Crankshaft NG 7/1/18
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always reproducible. 

I just installed Crankshaft NG,  and I'm having some strange issues that I'm not sure why I'm having. 
So NG is up and running,  I'm using a Waveshare 7 inch screen,  updated the config with the following as listed on the Waveshare website. 

max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1

This works in the original Crankshaft,  it should still work in NG correct?

But what I'm seeing with NG is that the touch display is very laggy,  almost not working most of the time. If I go into settings,  it takes several attempts at trying to push buttons for anything to happen.  It's almost like the touch input is not registering correctly. 

I also noticed when I plug my phone into the PI.   AA will start very slowly,  and it doesn't look right.  If off center/cut off on the right,  and I don't see the options that are normally at the bottom of the screen (maps,  apps etc).  

Any thoughts if I've done something wrong on installation,  or if this is a bug.
The original Crankshaft worked great,  so not sure why I'm having issues with NG.

chrisfromwa, 2018-07-03:
So I just tried hooking the PI up to my TV,  I'm not able to test out the touch responsiveness.   But when I plugged the phone in,   AA started pretty quickly and the buttons along the bottom are back.   Everything looked correct.

So it has to be something with my Waveshare screen,  even though it worked with the original Crankshaft.   Any ideas to what could be wrong,  and how to fix it?

hawkeyexp, 2018-07-04:
Hi that sounds a bit crazy but let's check something:

This sounds a bit crazy ... I have also a waveshare display optinal to my rpi org display
Try this please:

- boot into dev mode
- connect phone
- check if problem is still there
- disconnect phone
- type into shell: `sudo systemctl stop tap2wake`
- connect phone
- check again

ps: can you please let me know the output from `crankshaft usb identify` ? This is for another function
which i'm playing with :-)

chrisfromwa, 2018-07-04:
just gave it a try,  still doing the same thing.
Another I noticed,  is it supposed to automatically start Pandora when the phone is plugged in?  I don't recall it doing that before, but it does now for some reason. 

For the USB Identify thing,  how do I do that?  Let me know and I'll see what it says.

Hisma, 2018-07-04:
fyi, here is something I noticed with crankshaft-ng, it must use more processing power, because the power supply in my car that worked fine on the original crankshaft caused all sorts of strange anomolies on crankshaft-ng.  The touch screen would stop working, it would detect my phone as connected when it wasn't, sound wasn't working etc.  

I tried a more powerful power supply (one that supplies 2.5amps and wasn't shared w/ any other devices), and everything all of a sudden worked great.  

So I would look at your power supply if everything worked fine in original crankshaft vs crankshaft-ng.  Try to make sure you use a supply that supplies 2.5amps and is not shared w/ any other devices.

Maju3, 2018-07-04:
@Hisma 
I noticed exactly the same problems like you.
Touchscreen not working,
Message "phone connected" when it is disconnected.
No sound....

BUT i use a M3ATX psu which provides 6 ampere at 5 volt.
So i can say for sure that it is NOT related on the power supply.
Must be some bugs in csng i think.

Hisma, 2018-07-04:
@maju3

Actually I think you are right, though it appears to happen to me randomly.  I thought changing the power supply had fixed it, but it happened again in my car just recently.  I powered down and let it restart, and it was fixed. 

This will probably be difficult to solve bc it's not consistent.  Fwiw I have a hihiberry digi, not sure if hardware is common.  But this never happened on the old crankshaft.

Maju3, 2018-07-04:
Yes i tried today with crankshaft 0.2.2.
No problems.

Maju3, 2018-07-04:
Question:
How do you handle the phone connection to usb?
I mean do you connect the phone BEFORE booting the pi or after it has fully booted?

Maju3, 2018-07-04:
I think there is a problem with udev rules.
So csng does not really know if the phone is connected or not.
But this is only speculation.
But i noticed that when AA is running after lets say 5 minutes no button in AA gui works.
And when i disconnect the phone then mainscreen comes up and still shows " phone connected".
And even on mainscreen no button works.
It looks to me like the connection between phone and pi freezes.

Hisma, 2018-07-04:
I typically boot the unit prior to connecting my phone.  But it's not something I paid attention to when the failure occurred.

However, the first time it happened, I noticed that the screen stopped responding once I tried to play a song. So it makes me wonder if a failure occured with pulse audio or something related to the audio processing. The system still responded by closing and opening AA when I would disconnect and reconnect my phone. But I couldn't really do anything.

Maju3, 2018-07-04:
Yes same for me.

hawkeyexp, 2018-07-06:
Possible a bug is found - have to been tested again if still present in upcoming Pre2

Hisma, 2018-07-06:
@hawkeyexp please also see once again the bug with the touch screen and AA not responding when opening the backup camera view and then connecting the phone while the camera view is active.  It seems that the system cannot properly launch AA with the camera app running in the foreground.  I recreated this bug multiple times now.  At first I thought it was hw but I am now convinced this is a bug in the software.

hawkeyexp, 2018-07-06:
You are describing a default behaviour. The RPi Cam is controlled native and it's output is on highest layer of the gpu to make sure rearcam will be visible if triggered external - like commercial units do it, too.
I checked it out - openauto is launching but you won't see it behind the rearcam video. Also touch input is going to openauto.

Hisma, 2018-07-06:
The problem with me is that I cannot exit the camera and I have no touch input, so nothing works.  When I do a hard reboot, AA no longer launches anymore. I see 'phone connected' but openatuo doesn't launch.

I see you have a new revision coming out this weekend so I might try that first.  Regardless the camera feature as it stands is not much use to me until it can be auto triggered.  BTW, I wrote a script to create /tmp/backup_cam file, and it wouldn't display the camera.  I confirmed the file existed but no luck.  Is it all lowercase, and should it work from the shell or does it require autoapp service running?

hawkeyexp, 2018-07-06:
/tmp/backup_cam is not the backup trigger file.
With pre2 it is included to trigger via gpio upcoming this weekend

Hisma, 2018-07-06:
Excellent!  Thanks so much for your hard work!

chrisfromwa, 2018-07-09:
Hi,
I just installed the 7/8 release.  
I noticed that it's slightly less laggy,  but I'm still seeing issue where on the AA home screen,  everything seems off.   No navigation at the bottom,  I believe they are being cutoff,  and it's like the screen is too low and to the right as the clock, antenna icons are cut off. 

When I go into the settings area,  everything fits the screen fine.  But touch seems to be off,  like I have to press everything to the left for it to register.   For example,  to cancel out.  I have to press about half an inch to the left for it to let me out.  

Any ideas or settings that I could try?

chrisfromwa, 2018-07-10:
Anyone have any thoughts on my issue?  Am I the only one?

hawkeyexp, 2018-07-11:
try to set framebuffer manually if something changes:
`framebuffer_width=1024
framebuffer_height=600`

chrisfromwa, 2018-07-11:
I tried the 1024x600 settings,  doesn't seem to make any difference. 

The main crankshaft screen looks correct,  but if I go into AA it's still cutoff like the screenshot above.

chrisfromwa, 2018-07-11:
OK, I think I may have figured it out.
And this may be my fault why it was being cutoff. 

I looked up the screen I bought on amazon,  and it says the specs are actually 800x480.  Which is different than what I saw on the WaveShare site,  when I was pulling the specs from when I updated to NG.  

I haven't been able to hook my phone up yet,  but I have a feeling that was the issue.

hawkeyexp, 2018-07-12:
Thx for feedback.

closed 2018-07-12 by hawkeyexp