opencardev/crankshaft #140: [CSNG] Settings not saving

tehniemer, 2018-07-03:
In particular the DPI settings are not being saved when adjusted through the crankshaft interface. I have not tried directly editing the .ini file yet.

hawkeyexp, 2018-07-04:
Please make sure you pushed the save button after changing.
The system will save the ini file ONLY on a clean shutdown or reboot initiated by gui.
Shutdown / reboot via shell will interpreted as terminating and ignored to prevent from saving failed settings.

tehniemer, 2018-07-04:
I did press save and do a clean reboot. It still didn't save.

hawkeyexp, 2018-07-04:
I can't reproduce - all works fine
Have a look on /tmp/openauto.ini after saving settings in gui if file is written

unxpctederr, 2018-07-11:
My settings are not saving either.
/tmp/openauto.ini only contains settings for general, video, audio, input and bluetooth, but no settings for the Crankshaft NG options tab.

I have tried pressing the save button, allowing it to go back to the main screen, and then  selecting shutdown and waiting 60 seconds for automatic shutdown. However this does not save the settings under the Crankshaft NG tab (not sure about other tabs though).

hawkeyexp, 2018-07-11:
CS-NG settings are system related and not saved inside openauto.ini to keep a clean split between openauto and system.  For example volumes are stored in /boot/crankshaft/volume and capvolume.

unxpctederr, 2018-07-11:
I managed to edit those manually, however settings edited in the settings interface are still not saved.

closed 2018-07-23 by hawkeyexp