opencardev/crankshaft #141: General system settings on mainscreen

Maju3, 2018-07-04:
Just as suggestion.
What do you think about general system settings on mainscreen?
I mean like kodi or volumino do.
This way you have a gui where you can do settings for audio hardware like selecting which dac you use.
Or which audiooutput you use (3.5 jack or usb).
Or for the screen/pi shutdown timings like i suggested in an other post.
I just think that some gui for general settings makes it MUCH easier to config CS especially for people WITHOUT linux knowledge.

tehniemer, 2018-07-04:
This is a great idea, I would love to see this implemented.

hawkeyexp, 2018-07-04:
Thinking about some ways to release something - currently i'm extending the settings menu by a crankshaft page :-) So stay tuned...

Maju3, 2018-07-04:
Sounds good to me!
Thank you!

hawkeyexp, 2018-07-06:
Initial steps done in settings menu. It will be extended step by step.

closed 2018-07-06 by hawkeyexp