opencardev/crankshaft #142: [csng] hotspot mode not working

Grawa, 2018-07-04:
Hotspot not working even in dev or normal mode, doesn't create "crankshaft" network on channel 6 (default settings)

BOOT/CRANKSHAFT/CRANKSHAFT_ENV.SH
```
### Wifi Setup ###
# Hotspot (if enabled the wifi client is disabled and a hotspot is opened)
ENABLE_HOTSPOT=1
HOTSPOT_PSK=1234567890
```

tried with and without dev mode activated
```
### Maintenance / Initial Setup ###
# Start Crankshaft in dev mode to get network, shell and ssh access
# openauto won't be started automatically
DEV_MODE=1
# Start openauto in dev mode if enabled
DEV_MODE_APP=1
```

hawkeyexp, 2018-07-04:
I have checked it in my setup and it started without probs.

please do a quick check with `crankshaft hotspot status` and `crankshaft syslog` to get more info's whats going wrong and let me know

rubenverhoef, 2018-07-12:
I have the same problem.
`crankshaft hotspot status` returned:
`/usr/local/bin/crankshaft: line 1204: /tmp/dnsmasq.leases: No such file or directory`

hawkeyexp, 2018-07-13:
@rubenverhoef the output from crankshaft is not complete.
On top there must be two lines like:
`Hostapd: xxx`
`Dnsmasq: xxx`

But the failed line shows me that dnsmasq seems to be not started.

rubenverhoef, 2018-07-13:
The output of `Hostapd:` and `Dnsmasq:` gives nothing. Just the two words without anything behind it.

hawkeyexp, 2018-07-13:
Ok if no output services are not running. Check if `WIFI_COUNTRY=` is set in your config.

rubenverhoef, 2018-07-13:
`WIFI_COUNTRY` is default EN, that doesn't work and also NL (The Netherlands) does not work, same behaviour.
Like I said before, it is working when i have the LAN cable attached. So i think it should have something to do with it.

hawkeyexp, 2018-07-13:
There is no dep by lan cable...

let me know the output of following:
`sudo systemctl status hotspot`
`sudo systemctl status hostapd`
`sudo systemctl status dnsmasq`
`crankshaft syslog`

Any custom changes ?

rubenverhoef, 2018-07-13:
`sudo systemctl status hotspot` gives:

`● hotspot.service - Crankshaft Wifi Hotspot Service`
`   Loaded: loaded (/etc/systemd/system/hotspot.service; disabled; vendor preset: enabled)`
`  Active: inactive (dead)`

`sudo systemctl status hostapd`  gives:
`● hostapd.service - LSB: Advanced IEEE 802.11 management daemon`
`   Loaded: loaded (/etc/init.d/hostapd; generated; vendor preset: enabled)`
`   Active: inactive (dead)`
`     Docs: man:systemd-sysv-generator(8)`

`sudo systemctl status dnsmasq` gives:
`● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server`
`   Loaded: loaded (/lib/systemd/system/dnsmasq.service; disabled; vendor preset: enabled)`
`   Active: inactive (dead)`

`crankshaft syslog` gives:
https://pastebin.com/raw/vVMeaVtq

rubenverhoef, 2018-07-13:
Update: I reflashed my SD card and set the options:
`ENABLE_HOTSPOT=1`
`HOTSPOT_PSK=1234567890`

First boot in dev mode, working! reboot, not working anymore.
Booting with LAN cable attached, working. Booting without LAN cable attached, not working anymore.

hawkeyexp, 2018-08-03:
Please check again with a pre4 - will be public available on sunday - i have checked it some times and it was working fine. If problem is still there pre4 comes with a debug log option to collect system informations to add to issues to easier identify problems.

georgesreti, 2018-08-13:
Well I have been wondering what I was doing wrong for a while... CRANKSHAFT wifi network is not available when the ethernet cable is not attached with pre4, as described by rubenverhoef.

Even with the with a cable connected and then connected to the CRANKSHAFT network, android auto does not fire up. Not sure if it is a network interface mix up or me missing some setup? Is there anything special to do at this point?

I see pre5 is available now, I will try again and report back and try to get some logs.

hawkeyexp, 2018-09-01:
Tonight i tried again also including extracting my pi 3b+ from my media center to check if it is possible caused only on pi3b+ and working on pi3 but both are working like a charm.
No deps to ethernet etc. - tried with reboots - no bug - hotspots starts fine. 
sorry - closing cause no bug can be identified.

closed 2018-09-01 by hawkeyexp

Will-SCHOTT, 2020-10-02:
Hi All, no need to re-open that case, but just for information, I found a workaround to let hotspot activate by its own (needs to be done once only).

It consists in starting Samba service in settings (even though no network is up), then wait for it to confirm it's active, try again to connect to hotspot and... magic ;-)

Tested over 7 different CS setups and phones.