opencardev/crankshaft #143: Rotating screen does not work.

dinapappor, 2018-07-04:
Shorting pins 40 and 39 does not work (using a jumper). Adding lcd_rotate=2 to config.txt works briefly and then flips the image upside down again, like 5 or 10 seconds after boot.

I am using RPI 3B+ and official touch screen.

hawkeyexp, 2018-07-04:
Which Crankshaft version ? NG or Classic ?

dinapappor, 2018-07-04:
The latest one `Crankshaft OpenCarOS NG Pre1 2018-07-01`. Setting `FLIP_SCREEN=1` and adding `lcd_rotate=2` manually did seem to do the trick. Let me know if you need me to do any testing on my end.

hawkeyexp, 2018-07-05:
I have checked the gpio trigger and it works fine.
You do not need to set it in config.txt - during boot a service checks if it is set in crankshaft_env.sh or triggered by gpio and if yes it will setup the config.txt.
You can also check if system recognizes your gpio input with this commands:

`crankshaft gpio status` = shows your configured gpio's status
`crankshaft gpio all` = shows all gpio's status

Is the global ENABLE_GPIO enabled (set to 1) ? - if not 1 no gpio actions will be used.
PS: Make sure your gpio trigger stays closed to be set - the service checks it on every boot to revert flip if no longer needed.

hawkeyexp, 2018-07-23:
Closing - no feedback

closed 2018-07-23 by hawkeyexp