opencardev/crankshaft #145: [csng] rpi camera module causes major instability

Hisma, 2018-07-05:
- What phone do you have? What OS version?
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Using latest version of crankshaft ng, when I enable the camera module, I am having major instability.  AA no longer launches when I plug in my phone and the screen no longer responds to touch commands.  

I noticed that the major instability occurred when I rotated my screen 180 degrees in cranshaft_env.sh file.  This was necessary however due to the mounting of my camera.  

The moment I disabled the camera module using the 'crankshaft rpicam disable' command, all of the instability went away.  I would see if you can test how changing rotation affects stability.  

On a side note, raspivid works just fine.  I enabled it in raspi-config, so for my purposes, I may go that route as raspi-vid is easily scriptable.  Another thing that would be quite useful however, is having a gpio trigger option for turning on the camera.  This is a common use-case for a backup cam... you would want it to automatically turn on when going into reverse.  So please consider adding that feature as a gpio trigger.

Thank you for all your hard work.

hawkeyexp, 2018-07-05:
Hi Hisma, i checked it and can't reproduce any instability. I currently don't know if raspi-config does any additional setup which can cause the problem - raspi-config is not needed in CS-NG for cam setup.

About the gpio trigger to enable rearcam mode: this will come with next releases.
Currently it can be solved by creating a file called /tmp/rearcam_enabled. If this file is detected rearcam mode will be auto activated - if removed rearcam will be deactivated.

Hisma, 2018-07-05:
Can I give you a debug file?  I reproduced the problem after a reflash of the os.  I can also record a video showing what is happening.  I saw in another issue that you said debug mode can be turned on by placing a file on a USB stick right?

Hisma, 2018-07-05:
Actually, please close this issue!  I changed my usb power cable in my car, and everything just started working as normal.  

I wonder if other folks with stability issues have experimented with changing usb power cables.  Rpi seems to be very sensitive to clean power, and something with cs-ng, perhaps due to the additionally added features, the instability is more noticeable if not provided a clean connection.  Not just amount of power (large power supply), but also a clean signal path (quality cabling).  

Thanks!

closed 2018-07-05 by hawkeyexp

Hisma, 2018-07-05:
fyi... I re-created this fault again, after a fresh re-install and using a better cable.   So maybe it's not a power issue.  

Here's how I re-created it - 
- open the camera module, then display the rearview cam (w/ the 2 lines) 
- connect phone

See what happens when you do that.  Does android auto launch?  For me, it just continues to display the camera.  AA does not launch, and I am not able to press any more keys.  Even after a reboot, the problem of AA not launching persists.  I need to disable rpicam in order to restore AA functionality.

So please, can you confirm if you can run the backup camera and connect your phone to launch AA without issue?