opencardev/crankshaft #146: USB Microphone Gain Adjustment

Aaron407s, 2018-07-05:
Would it be possible to incorporate an entry in the config file that tweaks external USB microphone gain? I've noticed that I can't get voice commands to work with my microphone when I'm on gravel due to road noise, and often not on the highway. At slower speeds on paved roads it works great, so I'm wondering if a config.txt entry could be used to adjust the amixer/alsactl values for microphone gain.

Note: I realize that boosting the gain of the microphone may not fully fix my specific issue as the background noise would also be amplified, but I can't help but wonder if playing with the gain may have some benefits.

tehniemer, 2018-07-05:
Try moving the mic, I have mine on the steering column and I'm surprised at how well it picks up commands, even with the windows down.

chunky74, 2018-07-05:
Dammit that's an awesome idea - Mine is currently at the top of the A pillar - But I feel a move coming up !

hawkeyexp, 2018-07-06:
Planed to integrate...
Next Pre this weekend will have experimental integration to adjust system volume via gui in settings.
If this works fine it can be ported to mic vol, too.
[image]

hawkeyexp, 2018-07-06:
Also Mic volume control will be integrated in pre2 :-)

Maju3, 2018-07-07:
Hey showing system information and cpu temp is a great idea.
Well done!

hawkeyexp, 2018-07-09:
Released with CS-NG - please report back if it's working for you.

Hisma, 2018-07-09:
hey hawkeye, my usb microphone no longer works in the latest version (pre2).  It worked fine previously.  I noticed that when I tried to adjust mic gain (it showed 100% in the audio settings), even when I hit save, it always jumps to 0%.  So I think something is fubared where the mic gain is defaulting to 0% despite what the settings are.

Can you please confirm this?  I did not touch any of the mic wiring so I am led to believe this is software.

btw, reverse cam auto trigger works great :)

hawkeyexp, 2018-07-09:
Hm, in my case it works fine with my usb mic [()](https://www.amazon.de/Seacue-Omnidirektionaler-Kondensator-Interviews-Netzwerksingen/dp/B071171DBP/ref=sr_1_3?ie=UTF8&qid=1531116827&sr=8-3&keywords=usb+microphone).
If you don't change manually settings it works like before. Only after changing settings it will be touched.
How about system volume control ? Is this working for you ?

hawkeyexp, 2018-07-09:
Can you also check the alsamixer if capture is possible muted ?

Maju3, 2018-07-09:
I have tested the mic and master volume.
All working here.
BTW i use a usb dac and have disabled the 3.5 jack in config.txt.
#dtparam=audio=on

Aaron407s, 2018-07-09:
Thanks for hitting the ground running on this one, it's greatly appreciated. I'll give it a go within the next couple of days and report back. A loose subframe bolt will need my priority tonight, though :-)

hawkeyexp, 2018-07-09:
A general note:
#dtparam=audio=on to disable internal audio is a good idea cause it prevents from problems if more than one audio device is present. The controls are working with the system default (master) device.
With the management tool i have done same thing during setup of hifiberry dac's cause onboard audio is not longer needed in those situations.

Aaron407s, 2018-07-11:
I tried it out tonight, and everything works great. Being able to boost the audio a bit with the master audio control is fantastic as I have my HDMI output through a converter to a ground loop isolator that drops the level a bit. Well, it worked great after I realized that the dtparam=audio=on also affects HDMI audio. The microphone level setting also worked fine and was retained after reboots. I did a fresh install and tweaked the config.txt and crankshaft_env.sh files as necessary.

Thanks for incorporating this!