opencardev/crankshaft #148: [csng] Display blanking

smartroad, 2018-07-06:
- What phone do you have? What OS version?
Nokia 8, Android 8.1, Pi2, Official Touchscreen

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Use device normally and after a while the screen will turn blank - touching screen brings it back on, then turns off after a few minutes. Intermittent bug.

- Please provide any further information that you might find helpful.
Doesn't happen every time but have had it once where it turned off, touch screen, turned on and stayed on. Last night it kept turning off every few minutes (not sure exact length of time I was driving ;)

hawkeyexp, 2018-07-06:
Solved with next Pre

closed 2018-07-06 by hawkeyexp

smartroad, 2018-07-06:
Awesome!