opencardev/crankshaft #150: [csng] Touch screen unresponsive

smartroad, 2018-07-06:
What phone do you have? What OS version?
Nokia 8, Android 8.1, Pi2, Official Screen

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
Sometimes the system will not respond to touch inputs

- Please provide any further information that you might find helpful.
When this happens I have been able to set Waze to start navigating but then the touch becomes unresponsive. I only noticed first time as I was unable to move the map around. I also noticed that at the same time the sound stops working, so could be related to #149 or just a coincidence.

hawkeyexp, 2018-07-06:
Possible a bug is found - have to been tested again if still present in upcoming Pre2

smartroad, 2018-07-06:
Great will test once out :)

smartroad, 2018-07-07:
As an extra note, discovered it stops responding when the phone is connected before crankshaft has loaded/booted

Maju3, 2018-07-09:
In my case solved with csng pre2

smartroad, 2018-07-09:
Solved with ng pre2 here as well 😁

closed 2018-07-09 by smartroad

hawkeyexp, 2018-07-09:
perfect