opencardev/crankshaft #151: Pulseaudio equalizer

Maju3, 2018-07-06:
Just another suggestion.
In AA gui the right button on bottom for now just opens a black screen with a button called "return to openauto".
I think this screen would be the perfect place for sound control with pulseaudio equalizer.
This way you could set sound presets and volume and csng could completelly replace a car stereo unit.

smartroad, 2018-07-07:
Just realised my #152 is very similar to this request :grin:

Hisma, 2018-07-11:
Just a suggestion, but I would make a feature like this optional as it would likely add to the slow boot-up time we already experience.  I have an external DSP so I would personally have no need for an internal EQ... I just need my EQ to be flat at all times :).

Making it selectable from crankshaft_env.sh would be preferable.

Maju3, 2018-07-15:
Yes i agree.
For people who use external DSPs a equalizer makes no sense.
About that a setting(eq on/off) in crankshaft_env.sh or maybe in the homescreen gui would be great.

Aaron407s, 2018-07-18:
I believe that button is generally used for vehicle diagnostics (I believe it's a gauge icon). People such as myself trying to get OBD2AA working with Torque leverage this, so hopefully the basic functionality can still be maintained if needed, e.g., through a config setting.

hawkeyexp, 2018-07-23:
No equilizer will be released. The fight with pulseaudio is to hard - sorry.

closed 2018-07-23 by hawkeyexp

Hisma, 2018-07-23:
an alternative which to me is just as effective (though not as fancy) is to use an equalizer on your phone.  Viper4Android works really well, provided you have root.  Since all AA audio is routed through your phone, and the pi is just a pass-through device, having the equalizer on your phone achieves the same goal you desire.