opencardev/crankshaft #154: Csng, or cs, no bass during music playback.

pierre2013, 2018-07-07:
- What phone do you have? What OS version? P10 lite, and p10, and doogee x7, android 7, 8, and 6. I tried everything I could get my hands on.
- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
The problem is always there
- Please provide any further information that you might find helpful.

Music playback is completely void of any bass. Then google assistant speaks, and the sound is rich, even the music that continues, for about 2 seconds, and then it returns to the same annoying anemic music.

I have tried this on the built in jack, usb DAC, and also an i2c DAC. 

When I play back music from a kodi install, everything is good.

tehniemer, 2018-07-07:
I've noticed similar results. It seems a little better with the most recent pulse audio settings with a HiFiberry DAC+ and pulse equalizer installed, but could still be improved.

Maju3, 2018-07-08:
As far as i know Kodi uses its own soundserver instead of Pulseaudio.

I think it is not possible but what about to use the kodi soundserver(soundengine)for crankshaft?

Maju3, 2018-07-08:
Maybe a dumb idea from a noob:)
But i just ask:)

pierre2013, 2018-07-09:
So I think what the problem is, is that the speech is playing on the "music output", and vice versa.

htruong, 2018-07-13:
@pierre2013 Do you mind borrowing another phone to try? We seem to have a lot of troubles with Huawei branded phone for some strange reason.

pierre2013, 2018-07-15:
I have tried to borrow one, but the issue is that Namibia is not supported yet, so I have to jump through some hoops to get AA to even install. People don't like that...

Anycase, I realised on Saturday that I might be on the money when I was using a bluetooth speaker, and received a message. The music went "thin" for some seconds, and then recovered to some full rich sound again.

So with CS, it seems like its just the wrong channels playing...

pierre2013, 2018-07-22:
My p10 lite just updated to Android 8.

I will download the newest release of CS NG, and test it. I will report back soon.

pierre2013, 2018-08-04:
Hallo guys.

Sorry for taking so  long to give feedback... Life happened..

hawkeyexp, 2018-08-29:
Some tweaks will be added on next release - closing.

closed 2018-08-29 by hawkeyexp