opencardev/crankshaft #156: [csng] Backup Camera flickers while AA is active

Hisma, 2018-07-09:
Once again, great work integrating the backup camera auto-trigger function!  I am using it and I can report that it does work.  However, I notice that when AA is active, the screen tends to flicker back and forth between the backup camera view and the AA display.  At first I assumed this was a hardware issue, perhaps chattering as I am using a relay to trigger the gpio pin, but I noticed I have no flicker at all outside of AA with my phone disconnected.

The amount of flicker is not consistent.  Sometimes it flickers a lot, other times only a little.  But it only happens with AA active.  It is almost as though the OS is trying to force AA into the foreground but the camera is overriding it, so the two apps are "fighting" each other.

hawkeyexp, 2018-07-09:
You have to set omx layer index in settings from 2 to 1 :-)

Hisma, 2018-07-09:
I can confirm that fixed it!   However, I want to say, for some reason my settings are not persisting after reboot.  I take care to make sure I do a clean shutdown from the GUI.

Can I edit the openauto.ini file instead?  The settings not saving is definitely a separate issue, but until that gets fixed, is there some way I can force this setting to stay from the cli?

I suppose you can close this issue, but I can only temporarily apply the fix.  It needs to be re-applied on every start-up.

hawkeyexp, 2018-07-09:
About settings i will have a deeper look if there is a constellation which allows fails during writing settings - if there is a chance for fails i will address it for next pre..

closed 2018-07-09 by hawkeyexp

hawkeyexp, 2018-07-09:
Seems overlay index is missing during save. This is related to the core openauto. Will try to fix it and do a pull request to f1x. So it was a misunderstanding - only one param is not saved not full settings.

hawkeyexp, 2018-07-09:
Fix done for next pre - you can mod the file manually until next pre is released