opencardev/crankshaft #157: [CSNG] Restore custom wallpaper(s)

smartroad, 2018-07-09:
Just upgraded to NG2 via USB and it went off without any issues 😄 

One thing I did notice, it didn't backup or restore my custom wallpapers 😞

hawkeyexp, 2018-07-09:
Perfect if upgrade caused no problems.

About wallpapers - i forgot to include it to backup / restore :-(
I will add it to Pre3. But it will start to work on Pre3 to xxx.

Maju3, 2018-07-09:
Where do i change the wallpaper?
In old crankshaft versions it was in:
Boot/crankshaft/wallpaper.png

hawkeyexp, 2018-07-09:
The wallpapers are still located on /boot/crankshaft. If the wallpapers are missing there they will be placed during boot to prevent from system fails.

Maju3, 2018-07-09:
Ah okay i understand.
It is just that i have searched in boot/crankshaft on my sd card.
Directly after flashing the image to the card.
Without puting the card into the pi.
Tomorrow i take the card out of the pi and take a look again.
Thank you!

commit 38be547a (2018-07-12): [script] Fix #157
closed 2018-07-12 by hawkeyexp