opencardev/crankshaft #158: Google assistant cut offs/delay

Maju3, 2018-07-09:
When i use google maps then sometimes the voice is cutted off or delayed.
I try to explain:
E.g. Google assistant should say " drive on bakerstreeet for 22km".
Then it says " drive on baker".
Then music plays.
Then after 2 seconds music pauses and the assistant says "street for 22km".

Next to this i have noticed that SOMETIMES the assistant voice crackles and stucks.
Can anyone confirm this?

tehniemer, 2018-07-09:
I have noticed this behavior as well, although I'm not sure I have on CSNG as I don't use navigation too often.

smartroad, 2018-07-09:
Just used NG2 for two 2 hour journeys and Google maps was announcing okay for me 😕

Maju3, 2018-07-09:
Yes for me this issue happens mostly on first boot of pi.
Then when i unplug the phone and replug it google assistant talks without crackle or delay.
BTW normally i connect my phone BEFORE i start my car/pi.
Maybe this is the reason.
I do more tests tomorrow and try what happens when i connect the phone after the pi is booted...

hawkeyexp, 2018-07-09:
I can understand your way that you enter the car connect your phone without waiting until system has started. But try the clean way and connect after system has started for now.

Maju3, 2018-07-10:
Okay today i have connected the phone AFTER the system has started.
No voice stuttering.
But cause this issue happens only sometimes i need to test more.

Maju3, 2018-07-10:
Btw i think crankshaft initiates the phone after the system is completelly started.
About that it schouldn't make a difference if i connect the phone BEFORE starting the pi or after it is fully booted.
Or?

hawkeyexp, 2018-07-10:
Partitially your are right. In the background more things are happening. For example the notice phone connected in display is triggered by udev which will detect the device by rules and triggers /opt/crankshaft/usb_action.sh. But while starting android auto some changes are happening on the android device - the device and vendor id will change caused by aa to init the communication etc. so all is a bit critical and the recommended way is to connect if system is up and running.
Also during boot more resources are needed while services are starting etc. etc. and the optimal performance of the system is reached if all start processes are done.

Maju3, 2018-07-10:
Okay i understand:)
next days i keep on connecting the phone AFTER system start.
And i check for navigation voice stutter;)
Thank you!

hawkeyexp, 2018-07-12:
I think we can close for now and have an eye on it

closed 2018-07-12 by hawkeyexp