opencardev/crankshaft #160: [CSNG] No video with Kodi in EGL and crankshaft background in X11 mode

bttf92, 2018-07-10:
- What phone do you have? What OS version?
Moto G5 Plus with Android 7.0

- Steps to reproduce the bug, if you're planning to report a bug. Please indicate whether the bug is always repoducible.
If running in EGL mode this happens when playing a video:
[image].

In X11 mode, instead this happens:
[image]

- Please provide any further information that you might find helpful.
It's the same in dev mode.

It's a crankshaft or kodi issue?

hawkeyexp, 2018-07-10:
In EGL this is normal - overlays only works correct in X11. About the background: can be solved by skin using a black background behind Video.

But note: kodi is still unsupported by CS.

bttf92, 2018-07-10:
Ok, thanks! Will try! 👍

lockeandev, 2018-07-10:
I know its unsupported, but is there a resource anywhere outlining how to get kodi installed alongside crankshaft/AA? I see a few people doing it but can't find instructions for how to do it anywhere.