opencardev/crankshaft #161: [CSNG] Boot time slower than CS Classic?

pigr8, 2018-07-10:
Hi @hawkeyexp , is boot time (not in devmode) slower by design in the NG builds? comparing to classic it takes at least double the time to launch and see the AA interface.

I noticed that saving settings in AA is slower too, in classic the settings pannel is instantly hidden but in NG builds takes a bit, even entering settings pannel is slower.

Maju3, 2018-07-11:
Same for me.
Boot time of csng takes much longer as classic cs.

hawkeyexp, 2018-07-11:
@pigr8 let's have a look  on it...
Boot system until gui has reached - wait > 60 seconds.
Now the file /boot/crankshaft/startsequence.svg is created which shows startup sequence of systemd.
Add this file here. (Grab it manually from sdcard or copy it after reboot in devmode - must be done after start < 60 seconds otherwhise it will be overwritten).

pigr8, 2018-07-11:
Sometimes after power on it takes as long as 50 sec to show the AA gui (attached is a startsequence.svg after one of those boot), it needs about 36 sec to show the splash screen and ~14 from the splash to gui.

If i delete startsequence.svg and boot, it takes 9 sec to show the splash screen and ~23 to show the gui. so 32 sec total from power on to AA.. in this case if i wait <60 sec (and let linux create the file) and cycle a poweroff/poweron it needs again ~51 sec.

Even in the fast boot scenario it takes a lot comparing to Classic.. i tried to wipe the sdcard (Samsung EVO Plus U1 class) and reflash the F2E085FC, it boots really faster: after poweron in needs 6 sec to sho the first splash screen (grey with 3 white dots), after 3 sec the crankshaft splash shows up and after other 3 sec AA gui boots,  ~12 sec total.

All the tests are made in non-DEV mode.

[startsequences.zip](https://github.com/opencardev/crankshaft/files/2183294/startsequences.zip)

hawkeyexp, 2018-07-11:
Currently i can't reproduce - i always have ~30 seconds boot time to gui with a stable card.
The file startsequence.svg can't cause any change of boot time cause it's only a file created after boot that's all.
The content's of the log shows really long time for dev-mmcblk0p2.device so it could be part of the effect.

Generally: system is atm in pre state and not high optimized for boot times.
With later builds this will also be focused and optimized for stable releases (nobody loves to wait :-).
Let's first fix possible critical problems and extend the features to wanted level and after that fine tuning :-)

hawkeyexp, 2018-07-11:
Sure boot times will be optimized - but also the system got some things more which needs a bit of time - for example the whole usb-system for flashing etc.

pigr8, 2018-07-11:
Perfect, i just wanted to be sure that nothing was wrong my side, i was using daily the latest classic build and wanted to switch to see the improvements but as you said it has to get optimized so no big deal :) i'm gonna of course test future builds to see when i can jump to NG.

hawkeyexp, 2018-07-13:
First tunings will be available with Pre3 upcoming this weekend. ~ 20 sec in my case instead of 30sec.
I will close this issue cause i will have an eye on it and if possible optimizations will be included.

closed 2018-07-13 by hawkeyexp